stereolabs/zed-ros-wrapper #237: "CAMERA NOT DETECTED" message after running the ZED Calibration tool

hfvre, 2018-07-23:
Hi,

I'm using a ZED Camera on a Jetson TX2 with ubuntu 16.04 LTS. 

I have been working with my ZED for a few weeks without any problem. However the distances in a point cloud created by a ROS node I wrote were not correct so I used the ZED Calibration tool to calibrate my camera. The calibration was successful but since then, the camera is not detected anymore by my node or any of the ZED samples/tools. 

I tried to update the SDK to the latest version (2.5) but it didn't do anything

Is there a way to fix this issue ?

Thank you for your help

ravikt, 2018-08-03:
@hfvre Try installing the zed ros wrapper again. Also, if you have saved the previous calibration file then try to use that.

Myzhar, 2018-09-08:
@hfvre have you solved? Can the issue be closed?

hfvre, 2018-09-09:
Hi,

I'm sorry for the late answer. 
Unfortunately the solution suggested by ravikt didn't solve the issue.
After trying several times on different computers, I had to buy a new ZED camera. Everything is working fine now.

You can close this issue.

Thanks

closed 2018-09-09 by Myzhar