stereolabs/zed-ros-wrapper #240: Compilation Timing Error with reset_tracking.h

acwatkins, 2018-08-01:
I am building a docker container and it fails periodically on first build with this error.

```
In file included from /ros_catkin_ws/src/zed-ros-wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:20:0:
/ros_catkin_ws/src/zed-ros-wrapper/src/nodelet/include/zed_wrapper_nodelet.hpp:40:40: fatal error: zed_wrapper/reset_tracking.h: No such file or directory
compilation terminated.
```

This seems like a timing issue with the message generation. It fails on a clean build probably 1/2 of the time.

Myzhar, 2018-08-01:
It seems that it searches for included file before it is generated by ROS.
I compiled more than a time from scratch deleting `devel` and `build` folders in the catkin workspace and I never had this problem.
I'm going to test to see if something changed.

acwatkins, 2018-08-01:
It is timing related, so CPU speed and some other factors play into it. I only see it on amd64 builds in CI and not arm64 for instance.

I have had this issue on my projects before. I added something like my_msgs_generate_messages_cpp to the add_dependencies line in CMakeLists.txt and it fixed it.

https://answers.ros.org/question/188982/message-headers-wont-build-first/

commit 7700134a (2018-08-02): Fixed #240
closed 2018-08-06 by Myzhar