stereolabs/zed-ros-wrapper #243: Is zed mini IMU and camera hardware synchronized?

gajena, 2018-08-05:


Myzhar, 2018-08-06:
The timestamp associated to each IMU message is the timestamp of the latest camera frame

zelk, 2018-08-13:
Do you mean that the timestamps of the IMU ticks only at the same rate as the camera?

gajena, 2018-08-13:
@zelk I am asking that IMU value and image data that are taken together should have same timestamp!

gajena, 2018-08-13:
@zelk thanks to you for information!
I have one more question!
Can I use zedsdk only and only for getting synced IMU and image data ?

Myzhar, 2018-08-13:
Surely. You can follow one of the example in the repoditory
https://github.com/stereolabs/zed-examples
and retrieve IMU data like in the ROS wrapper:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1101

Myzhar, 2018-08-15:
Yes, I missed the part where you were asking about using the SDK without CUDA.
The ZED SDK is deeply based on CUDA and cannot be used on system without a Nvidia GPU.

Myzhar, 2018-09-08:
I close the issue but you can continue to pay comments if you need further help

closed 2018-09-08 by Myzhar

rhklite, 2018-09-28:
> We could use the "real" timestamp, but TF2 does not like it and generates a lot of timing errors.

I'm running into some issue with imu/camera time stamping when i'm feeding the two into a image processing pipeline. Is there a way to force the IMU to use the ""real" timestamp?

Myzhar, 2018-09-28:
Yes, you can change this line:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1355
replacing `mFrameTimestamp` with `ros::Time::now()`

Let us know if it works for you.