stereolabs/zed-ros-wrapper #247: Why imu_link and not zed_imu_link?

Affonso-Gui, 2018-08-08:
On recent updates we got a new `imu_link` in [zedm.urdf](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zedm.urdf#L70-L77), which is good news.

The problem is that this `imu_link` is causing name clashing with the `imu_link` I already had on my robot.

Considering that all of the other links have a `zed_` prefix, my question is: why is it named `imu_link` instead of `zed_imu_link`? I believe that the latter would be both easier to understand and easier to include in robot urdf.

Myzhar, 2018-08-08:
The default name for IMU frames is defined in 
http://www.ros.org/reps/rep-0145.html
If "imu_link" generates conflicts with your system you can change the frame name here:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed_camera.launch
and modify the urdf replacing "imu_link" with the new name you set

Affonso-Gui, 2018-08-08:
Thanks for the reply.
The argument `imu_frame` was not being passed to the main node, so I just add it in https://github.com/stereolabs/zed-ros-wrapper/pull/248.

Best regards.

closed 2018-08-08 by Affonso-Gui