stereolabs/zed-ros-wrapper #249: ZED not working after boot but replugging it helps on J120 TX2

ArkadiuszNiemiec, 2018-08-09:
I have a TX2+J120 carrier board with Logitech USB Camera connected to OTG Micro USB, LTE USB Modem connected to USB3 and ZED connected to the second USB3.
Everything work but ZED camera requires me to remove and plug again after every reboot. Do you know what may be the cause of this?

Full dmesg log: https://gist.github.com/ArkadiuszNiemiec/88a0116fc53319efc49b205a77b166e3

At around 120 seconds you can see me replugging the ZED camera.

`[    0.000000] Booting Linux on physical CPU 0x100
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.38-tegra (auvidea@auvidea-T5600) (gcc version 5.3.1 20160113 (Linaro GCC 5.3-2016.02) ) #1 SMP PREEMPT Mon Apr 30 16:06:06 CEST 2018
[    0.000000] Boot CPU: AArch64 Processor [411fd073]
[    0.000000] earlycon: Early serial console at MMIO32 0x3100000 (options '')
[    0.000000] bootconsole [uart0] enabled
[    0.000000] Reserved memory: initialized node ramoops_carveout, compatible id nvidia,ramoops
[    0.000000] cma: Reserved 64 MiB at 0x00000000fc000000
[    0.000000] On node 0 totalpages: 2052096
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 519680 pages, LIFO batch:31
[    0.000000]   Normal zone: 23968 pages used for memmap
[    0.000000]   Normal zone: 1532416 pages, LIFO batch:31
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 17 pages/cpu @ffffffc1f5f9a000 s31360 r8192 d30080 u69632
[    0.000000] pcpu-alloc: s31360 r8192 d30080 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 2019936
[    0.000000] Kernel command line: root=/dev/mmcblk0p1 rw rootwait console=ttyS0,115200n8 console=tty0 OS=l4t fbcon=map:0 net.ifnames=0 memtype=0 video=tegrafb no_console_suspend=1 earlycon=uart8250,mmio32,0x03100000 nvdumper_reserved=0x2772e0000 gpt tegraid=18.1.2.0.0 tegra_keep_boot_clocks maxcpus=6 boot.slot_suffix= boot.ratchetvalues=0.2.1 androidboot.serialno=0321417073610 bl_prof_dataptr=0x10000@0x277040000 sdhci_tegra.en_boot_part_access=1 root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Memory: 7976360K/8208384K available (11112K kernel code, 1764K rwdata, 5704K rodata, 1144K init, 684K bss, 166488K reserved, 65536K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vmalloc : 0xffffff8000000000 - 0xffffffbdbfff0000   (   246 GB)
                   vmemmap : 0xffffffbdc0000000 - 0xffffffbfc0000000   (     8 GB maximum)
                             0xffffffbdc2000000 - 0xffffffbdc9da0000   (   125 MB actual)
                   fixed   : 0xffffffbffa7fd000 - 0xffffffbffac00000   (  4108 KB)
                   PCI I/O : 0xffffffbffae00000 - 0xffffffbffbe00000   (    16 MB)
                   modules : 0xffffffbffc000000 - 0xffffffc000000000   (    64 MB)
                   memory  : 0xffffffc000000000 - 0xffffffc1f6800000   (  8040 MB)
                     .init : 0xffffffc0010ed000 - 0xffffffc00120b000   (  1144 KB)
                     .text : 0xffffffc000080000 - 0xffffffc0010ed000   ( 16820 KB)
                     .data : 0xffffffc00122c000 - 0xffffffc0013e5160   (  1765 KB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 64.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=6.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=64, nr_cpu_ids=6
[    0.000000] NR_IRQS:64 nr_irqs:64 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] Architected cp15 timer(s) running at 31.25MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xe6a171046, max_idle_ns: 881590405314 ns
[    0.000002] sched_clock: 56 bits at 31MHz, resolution 32ns, wraps every 4398046511088ns
[    0.009816] Console: colour dummy device 80x25
[    0.014475] console [tty0] enabled
[    0.018031] bootconsole [uart0] disabled
[    0.022133] kmemleak: Kernel memory leak detector disabled
[    0.022145] Calibrating delay loop (skipped), value calculated using timer frequency.. 62.50 BogoMIPS (lpj=125000)
[    0.022156] pid_max: default: 32768 minimum: 301
[    0.022197] Security Framework initialized
[    0.022226] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.022236] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes)
[    0.022587] Initializing cgroup subsys freezer
[    0.022598] Initializing cgroup subsys debug
[    0.022732] CPU0 ipc=752
[    0.022741] CPU1 ipc=1024
[... 867 lines cut ...]
[    9.092344] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work ignore firmware MBOX_CMD_DEC_SSPI_CLOCK request
[    9.358823] scsi 2:0:0:0: Direct-Access              SD Storage       0000 PQ: 0 ANSI: 2
[    9.372533] scsi 2:0:0:1: CD-ROM                     Mass Storage     0000 PQ: 0 ANSI: 2
[    9.384210] sd 2:0:0:0: [sda] Unit Not Ready
[    9.388610] sd 2:0:0:0: [sda] Sense Key : 0x5 [current] 
[    9.394037] sd 2:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[    9.409144] sd 2:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[    9.417687] sd 2:0:0:0: [sda] Sense Key : 0x5 [current] 
[    9.423027] sd 2:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[    9.428593] sd 2:0:0:0: [sda] Write Protect is off
[    9.433484] sd 2:0:0:0: [sda] Mode Sense: 0f 00 00 00
[    9.439048] sd 2:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    9.457629] sd 2:0:0:0: [sda] Unit Not Ready
[    9.461964] sd 2:0:0:0: [sda] Sense Key : 0x5 [current] 
[    9.467312] sd 2:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[    9.480216] sd 2:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x00 driverbyte=0x08
[    9.488776] sd 2:0:0:0: [sda] Sense Key : 0x5 [current] 
[    9.494115] sd 2:0:0:0: [sda] ASC=0x20 ASCQ=0x0 
[    9.500972] sd 2:0:0:0: [sda] Attached SCSI removable disk
[    9.766477] fuse init (API version 7.23)
[    9.842826] usb 1-1: feature bit otg_vbus_off set
[    9.847627] usb 1-1: New USB device found, idVendor=046d, idProduct=082d
[    9.854415] usb 1-1: New USB device strings: Mfr=0, Product=2, SerialNumber=1
[    9.861612] usb 1-1: Product: HD Pro Webcam C920
[    9.866295] usb 1-1: SerialNumber: 0152B`

plieningerweb, 2018-08-14:
Porblem was solved for me after message "CAMERA DETECTION ISSUE" by flipping the usb -C connector 180degree and plugging it in again.

BUT YOU DEFINITELY SHOULD FIX THAT IN THE FIMRWARE OR SOMEWERE OR AT LEAST GIVE CLEAR INSTRUCTIONS. FOund it only accidentially in the zed diagnotics tool where it says "flip or use usb3.0 port"

molinadavid, 2018-08-16:
I have the exact same issue on a Jetson tx2, on boot up the camera does not work, but re connecting it solves the problem, however some time it takes more than one try.

ArkadiuszNiemiec, 2018-08-21:
@plieningerweb it's true but it's also not the problem I described. 
@molinadavid what carrier board are you using? Dev board?

molinadavid, 2018-08-21:
@ArkadiuszNiemiec I am using a Jetson TX2 Developer Kit.

ArkadiuszNiemiec, 2018-08-21:
@molinadavid thanks, I was worried that it is the problem of my J120 carrier board.

molinadavid, 2018-08-21:
@ArkadiuszNiemiec I can also confirm that is not a specific board, I have 7 different kits all present the same problem, also can confirm that this problem is only on the ZED-mini, I have also ZED normal type (the big one) and this problem does not happen.

jonpol01, 2018-08-22:
This problem occurs too in a regular windows pc. Have tried both firmware 14xx & 15xx, but still, no camera detected and replugging the camera sometimes fixes it.

ArkadiuszNiemiec, 2018-08-23:
@nesnes Good news, thank you! Please keep us updated if you will find a fix.
Do you know if it was fixed by the separate power rail or the USB Hub chipset?

Myzhar, 2018-08-23:
@ArkadiuszNiemiec Removing the power cord from the hub the problem remains. That is a good USB3 HUB for ZED because it has a very short USB cable

jonpol01, 2018-08-24:
@nesnes 
Actually using a self powered USB hub was our first workaround on it too.. But we found it very unpractical solution on our closed, compact mobile environment eg. Drone, Robot.

However, we appreciate your efforts on it, and are waiting for a real fix.

Wagdi-Benyaala, 2018-09-03:
Hi @Myzhar,

Is there any update on this issue ? I also managed to reproduce it (jetson tx2 dev kit + ZED-M, classical ZED works fine at boot) and unfortunately the powered usb 3.0 is not a working solution for me :( !

jeff-o, 2018-09-10:
We found we were able to adjust the "USB Power Delay" BIOS setting on our mini-ITX motherboard to 5 seconds to solve some of these issues.

jonpol01, 2018-09-11:
@Myzhar 
Any news on this issue? we are on the verge of dropping the use of zed-mini and ask for refunds for 30units we bought in bulk from stereolabs.

Myzhar, 2018-09-11:
@jonpol01 We are discussing with the Nvidia Embedded department to find a solution for this problem (maybe a kernel patch). We will update as soon as possible

jonpol01, 2018-10-04:
@Myzhar any updates?

obraun-sl, 2018-10-04:
The bug is in process in NVIDIA since this is related to the Jetpack boot configuration. Sorry for the inconvenience but we don't have the full control on the Jetpack.
We are also looking at the jetpack kernel on our side to see if we can have a custom Jetpack to provide, but that would be best if it's official from NVIDIA.

jonpol01, 2018-10-04:
@obraun-sl @Myzhar 
Im sorry but, if you gonna backtrack our reports of this bug, this too occurs in a ordinary PC environment.
Its unfair to push the blame to Jetpack since its clear to us its Hardware Zed-mini problem related.

ArkadiuszNiemiec, 2018-10-04:
@jonpol01 I can confirm what you wrote. Ubuntu 18.04 on Mi Notebook Pro 15" - same problem but solving it on TX2 will be enough for me.

molinadavid, 2018-10-29:
@obraun-sl @Myzhar Any news regarding this? is getting really frustrating not having news or the fix to the problem, could you at least release a workaround for it? maybe a kernel patch or a custom Jetpack while the official comes out.

obraun-sl, 2018-10-29:
Hi,
We are currently working on a kernel patch. We have found that the problem does not happen on a TX1 , therefore we are looking at the difference to understand why it does not boot.

molinadavid, 2018-11-01:
By the way I can confirm that the same issue still true on the newest Xavier as well.

nominator, 2018-11-22:
any update on this? SDK 2.7.1 was released on 16th last week but there is no mention of this being fixed in the release notes. So I assume it is still up there. 

However, I would like to know from people here who have experienced this issue that does it reliably comes back after replugging? I am asking this because I need to start work on a project and as long as it reliably get detected by replugging I can continue my work until a proper fix is provided.

Thanks

ArkadiuszNiemiec, 2018-11-22:
> However, I would like to know from people here who have experienced this issue that does it reliably comes back after replugging? I am asking this because I need to start work on a project and as long as it reliably get detected by replugging I can continue my work until a proper fix is provided.

Yes, it comes back after replugging, no problem after. For development it's OK, we just cannot apply it in production devices.

nominator, 2018-11-22:
Thanks @ArkadiuszNiemiec. Big relief to know that :+1:

AdarshK1, 2018-12-01:
Any update on this issue?

molinadavid, 2019-01-11:
Since no official fix (or even updates on the issue) has been released I had success with a workaround that involves re building the TX2 kernel, decompiling/recompiling the device tree, removing the vbus regulator and triggering a script with a serviced to turn off and on the USB port on boot up.

For anyone interested on the rough steps to fix this this is the thread on the nvidia forum (USB Power Control)[https://devtalk.nvidia.com/default/topic/1042460/jetson-tx2/usb-power-control/post/5289809/#5289809].

ahmad-aljabali, 2019-03-19:
I fond that launching the Zed Explorer (which doesn't require CUDA) will fix this problem without disconnecting and reconnecting.

I'm using Ubuntu 16.04, on an X86_64 pc.
This simple bash script does the job for me so I can use it when the camera is inaccessible, but it doesn't solve the issue when multiple cameras are connected.
 
#!/bin/bash
"/usr/local/zed/tools/ZED Explorer" &
PID=`jobs -p`
sleep 5
kill $PID

adujardin, 2019-03-20:
The fix for the ZED-M detection has been integrated by NVIDIA into the [new JetPack 4.2](https://developer.nvidia.com/embedded/jetpack).
The TX2 can now reboot with the ZED-M plugged in and it will be detected, like the ZED.

A compatible ZED SDK installer has been [released](https://www.stereolabs.com/developers/release/2.7/#sdkdownloads_anchor) for the TX2.

jonpol01, 2019-03-22:
I wouldnt consider it a fix, since Jetpack 4.2 reflashes your tx2 system into ubuntu 18.04.
Its a slap in our face thats says "go upgrade your 16.04 dev codes to 18.04"

However we did a workaround with the help from Nvidia guys on the developer forums to fix this ugly Zed mini hardware fault.

btw, stereolabs developers sucks on support and addressing this issue publicly.

We are considering lawsuits for this false advertising. "that it works on Ubuntu linux(jetpack and Zed sdk version from summer 2018 to this date) without doing workarounds on the dtb(which we solved for you guys)"

ArkadiuszNiemiec, 2019-03-22:
@jonpol1 can you post a link to forums, where the problem was discussed?
It was said here that it's a hardware problem and the solution will be only a workaround. And thus we have dropped the project that was intended to use ZED Mini but I am interested in the discussion anyway.

jonpol01, 2019-03-22:
@ArkadiuszNiemiec 
https://github.com/stereolabs/zed-ros-wrapper/issues/249#issuecomment-453387119

obraun-sl, 2019-03-22:
@jonpol01 
Thanks for your feedback.

I understand you're not happy that the fix requires you to update your Jetson to the new Jetpack 4.2 (which is based on Ubuntu 18).
After discussion with the team and Nvidia, we can provide as well a patch to apply to the Jetpack 3.3 kernel (Ubuntu 16). It is a diff file (see attached) that you need to apply to the kernel sources. Once applied, you will need to rebuild the kernel and flash the Jetson with the new image. That way, you can keep using Jetpack 3.3 (and Ubuntu 16).

Nvidia will also backport the fix to JetPack 3.X, so that it does not require to rebuild the Jetpack.
Again, sorry if we didn't not provide a fix quickly after the issue was discovered but we hope the above solution will help.

[f8f9f49.zip](https://github.com/stereolabs/zed-ros-wrapper/files/2996555/f8f9f49.zip)

ArkadiuszNiemiec, 2019-03-31:
Can someone test it so I can close this issue? My TX2 is currently without a carrier board.

silent-code, 2019-05-03:
I have tested the current Jetpack 3.3 and can confirm that Jetpack 3.3 as of today, May 3rd 2019, does not fix this bug.

tomcattigerkkk, 2019-05-14:
You can try to add a rule in your system 
e.g. /etc/udev/rules.d/zedmini.rules  // the name of rule is not important
in this rule file, add a line 
`SUBSYSTEM=="usb", ATTR{idVendor}=="2b03", ATTR{idProduct}=="f681", MODE="0666"`
the two ids can be checked through lsusb. 
In my case, I do not need to replug the usb cable before each new start.

rhklite, 2019-06-13:
@tomcattigerkkk 
> You can try to add a rule in your system
> e.g. ` /etc/udev/rules.d/zedmini.rules` // the name of rule is not important
> in this rule file, add a line to identify the id of the ZED device
> `SUBSYSTEM=="usb", ATTR{idVendor}=="2b03", ATTR{idProduct}=="f681", MODE="0666"`
> the two ids can be checked through lsusb.
> After adding this file, restart the TX2.
> In my case, I do not need to replug the usb cable after each new start.

this fix did not work for me. i also tried the following variations:
`SUBSYSTEM=="usb", ATTRS{idVendor}=="2b03", ATTRS{idProduct}=="f681", MODE="0666"`
`SUBSYSTEM=="usb", ATTRS{idVendor}=="2b03", ATTRS{idProduct}=="f681", MODE:="0666"`
`SUBSYSTEM=="usb", ATTR{idVendor}=="2b03", ATTR{idProduct}=="f681", MODE:="0666"`

with reboot each time in between the changes.

is there anything that i might've missed?

```bash
ubuntu@tegra-ubuntu:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2b03:f681  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

obraun-sl, 2019-06-14:
@rhklite 
Please update to last Jetpack 4.2 to have the fix for the ZED Mini boot. It's not yet in previous jetpack.
(The rule file is only for the HID device which is already detected at boot on your jetson (2b03:f681)  )

cosama, 2019-07-23:
Is there any news of this being backported to earlier Jetpack versions? Will it be necessary to reflash the TX2 or can the fix be applied through a software upgrade?

Myzhar, 2019-07-24:
@cosama if the fix has been backported to older Jetpack versions a reflash is anyway required

annaborn, 2020-04-06:
@obraun-sl 
I have Jetson TX2 carrier board J140 with the Jetpack 4.3 (auvedia firmware). I still have the issue that was described here: if I run lsusb:

`lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 030: ID 2b03:f681  
Bus 001 Device 035: ID 046d:c050 Logitech, Inc. RX 250 Optical Mouse
Bus 001 Device 034: ID 046a:0023 Cherry GmbH CyMotion Master Linux Keyboard G230
Bus 001 Device 031: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/3p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=tegra-xusb/4p, 480M
    |__ Port 2: Dev 31, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 35, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 34, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 3: Dev 34, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 3: Dev 30, If 0, Class=Human Interface Device, Driver=usbhid, 12M`

I tried both usb ports, different cables and rotate the cable around. I tried also the cam on my notebook, it detects properly, but not on jetson board.

horverno, 2020-06-05:
there is a workaround for this issue: use an external supplied usb cable _(hope this information is useful for somebody)_

amineKourta, 2022-03-21:
Same here ZED mini Jetpack 4.6 need to replug it everytime to get it working is there a workaround for this ?

closed 2022-04-26 by github-actions[bot]

maxpolzin, 2022-08-08:
Issue observed on Xavier NX on Auvidea JNX30 carrier board with Jetpack 5.0.1 and ZEDSDK 3.7
Turning the cable around fixed the issue.

maxpolzin, 2022-11-09:
Also observed for the Seeedstudio A206 carrier with Jetpack 5.0.2 and ZEDSDK 3.8.