stereolabs/zed-ros-wrapper #253: TF_OLD_DATA warning on Jetson TX2

ravikt, 2018-08-16:
```
Warning: TF_OLD_DATA ignoring data from the past for frame zed_camera_center at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame zed_camera_center at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame zed_camera_center at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame zed_camera_center at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.53445e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.17/src/buffer_core.cpp
```

Hello All,
I am getting a warning message when I launch the zed_wrapper. This happens whenever I do rosbag record. Can anyone suggest how to fix this issue ? Any help will be appreciated.

Myzhar, 2018-08-16:
Does this happen while recording or while reproducing the rosbag?

You can find a discussion about the error here:
https://answers.ros.org/question/201948/tf_old_data-ignoring-data-from-the-past-for-frame-odom/

ravikt, 2018-08-16:
It happens while recording the rosbag. It's happening in real time

Myzhar, 2018-08-20:
The rosbag produced is valid?

ravikt, 2018-08-20:
Yes, the rosbag is valid

ravikt, 2018-08-24:
The issue exists even while using the camera with desktop machine.

Myzhar, 2018-08-24:
Sure. Desktop hard disk are surely not faster than TX2 ROM.

Myzhar, 2018-09-08:
I close the issue but you can continue to post comments if you need further help

closed 2018-09-08 by Myzhar