stereolabs/zed-ros-wrapper #254: ZED ros interrupts wireless mouse

ArtlyStyles, 2018-08-19:
Hi,

I am running ZED ROS wrapper on MIT-Racecar, a car-like robot, which uses jetson TX2. The ZED ros node is started by zed_wrapper/zed.launch. TX2 is connected to a monitor, and I am using a USB wireless mouse/keyboard on TX2. I found however, once ZED ros node was started, the wireless mouse became irresponsive. The delay was very long. Anyone knows why?

Thanks,

Myzhar, 2018-08-19:
Are ZED and the mouse dongle connected to the same USB3 port using and USB3 hub?

Myzhar, 2018-08-19:
First of all be sure that it is an USB3 hub, not USB2.
If you continue to have problems connect the ZED alone to the USB3 port and the other peripherals to the "miniUSB" port with an adapter.
Remember that the "miniUSB" port should not be used with the ZED because it is an USB2 port and ZED needs USB3 to get full FPS

ArtlyStyles, 2018-08-19:
I am using this USB hub https://www.amazon.com/gp/product/B00E6GX4BG/ref=ask_ql_qh_dp_hza which claims to be USB 3.0. On this hub, besides ZED, I have IMU, teleop with game pad (look like a Xbox remote controller ). I can definitely move the mouse/keyboard to miniUSB. But not sure if I can move IMU and teleop to USB 2.0

closed 2018-11-24 by Myzhar