stereolabs/zed-ros-wrapper #256: Can't run from an SVO file

jack-scott, 2018-08-21:
Hi,

I encountered an issue while trying to launch the zed ROS wrapper with an SVO file. The command i'm using to launch is: 

`roslaunch zed_wrapper zed.launch svo_file:=../Zed_data/720at60fps.svo `

The error I receive is;

```
ZED (Init) >> ERROR: Cannot load SVO file. Check that the path of the SVO is correct, avoid white spaces in filename, add ".svo" extension to filename.
[ INFO] [1534852370.610837839]: INVALID SVO FILE
```
The file definitely exists and the path is correct as I have used tab complete. I had a look for the source of the error however could not locate the file in which it is being thrown. Any ideas what might be happening here?

Appreciate your help,

Jack

Myzhar, 2018-08-21:
The syntax you are using indicates that you are using an old version of the wrapper.
Can you update to the latest master?

jack-scott, 2018-08-26:
Hi thanks for the reply. I have pulled an built the latest master, what syntax should i be using?

Myzhar, 2018-08-26:
roslaunch zed_wrapper zed.launch svo_file:=your_file.svo
to launch the node
roslaunch zed_display_rviz display.launch svo_file:=your_file.svo
to launch the node with Rviz visualization

jack-scott, 2018-08-26:
I just verified that i can play it back in the ZED explorer as well as make a mesh in ZEDfu. It appears to be valid

Myzhar, 2018-08-26:
Can you copy and past the command you are using? I use SVO files daily with no problem.
If you are using a relative path maybe that the wrapper does not resolve it in the correct way

jack-scott, 2018-08-29:
Alright I found my issue, you were right the wrapper was not resolving the relative path, i had to give the absolute path. Thanks for you help

closed 2018-09-08 by Myzhar