stereolabs/zed-ros-wrapper #260: Waiting for image

chijepark, 2018-08-23:
Hello 
I am new in YOLO and ROS
So I have a problem but I don't know how i can run corretly
anyone have same problem before ??
I am using TX2 board and zed camera.

when i type like this
$ roslaunch darknet_ros darknet_ros.launch

I got this message

... logging to /home/nvidia/.ros/log/66af50ec-a676-11e8-9e25-00044ba8064b/roslaunch-tegra-ubuntu-6722.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:37211/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions/camera_reading/name: /darknet_ros/chec...
 * /darknet_ros/config_path: /media/nvidia/nvm...
 * /darknet_ros/image_view/enable_console_output: True
 * /darknet_ros/image_view/enable_opencv: True
 * /darknet_ros/image_view/wait_key_delay: 1
 * /darknet_ros/publishers/bounding_boxes/latch: False
 * /darknet_ros/publishers/bounding_boxes/queue_size: 1
 * /darknet_ros/publishers/bounding_boxes/topic: /darknet_ros/boun...
 * /darknet_ros/publishers/detection_image/latch: True
 * /darknet_ros/publishers/detection_image/queue_size: 1
 * /darknet_ros/publishers/detection_image/topic: /darknet_ros/dete...
 * /darknet_ros/publishers/object_detector/latch: False
 * /darknet_ros/publishers/object_detector/queue_size: 1
 * /darknet_ros/publishers/object_detector/topic: /darknet_ros/foun...
 * /darknet_ros/subscribers/camera_reading/queue_size: 1
 * /darknet_ros/subscribers/camera_reading/topic: /camera/rgb/image...
 * /darknet_ros/weights_path: /media/nvidia/nvm...
 * /darknet_ros/yolo_model/config_file/name: yolov2-tiny.cfg
 * /darknet_ros/yolo_model/detection_classes/names: ['person', 'bicyc...
 * /darknet_ros/yolo_model/threshold/value: 0.3
 * /darknet_ros/yolo_model/weight_file/name: yolov2-tiny.weights
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

ROS_MASTER_URI=http://localhost:11311

process[darknet_ros-1]: started with pid [6739]
[ INFO] [1534991172.552492792]: [YoloObjectDetector] Node started.
[ INFO] [1534991172.561037444]: [YoloObjectDetector] Xserver is running.
[ INFO] [1534991172.565032961]: [YoloObjectDetector] init().
YOLO V3
layer     filters    size              input                output
    0 conv     16  3 x 3 / 1   416 x 416 x   3   ->   416 x 416 x  16  0.150 BFLOPs
    1 max          2 x 2 / 2   416 x 416 x  16   ->   208 x 208 x  16
    2 conv     32  3 x 3 / 1   208 x 208 x  16   ->   208 x 208 x  32  0.399 BFLOPs
    3 max          2 x 2 / 2   208 x 208 x  32   ->   104 x 104 x  32
    4 conv     64  3 x 3 / 1   104 x 104 x  32   ->   104 x 104 x  64  0.399 BFLOPs
    5 max          2 x 2 / 2   104 x 104 x  64   ->    52 x  52 x  64
    6 conv    128  3 x 3 / 1    52 x  52 x  64   ->    52 x  52 x 128  0.399 BFLOPs
    7 max          2 x 2 / 2    52 x  52 x 128   ->    26 x  26 x 128
    8 conv    256  3 x 3 / 1    26 x  26 x 128   ->    26 x  26 x 256  0.399 BFLOPs
    9 max          2 x 2 / 2    26 x  26 x 256   ->    13 x  13 x 256
   10 conv    512  3 x 3 / 1    13 x  13 x 256   ->    13 x  13 x 512  0.399 BFLOPs
   11 max          2 x 2 / 1    13 x  13 x 512   ->    13 x  13 x 512
   12 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024  1.595 BFLOPs
   13 conv    512  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x 512  1.595 BFLOPs
   14 conv    425  1 x 1 / 1    13 x  13 x 512   ->    13 x  13 x 425  0.074 BFLOPs
   15 detection
mask_scale: Using default '1.000000'
Loading weights from /media/nvidia/nvme0n1/yolo_chije/ws_yolo/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov2-tiny.weights...Done!
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.
Waiting for image.

should I make new node to publish image??

Myzhar, 2018-08-23:
This is not an issue due to ZED, but the solution is pretty simple:
1) Open a terminal and launch: 
`$ roscore`
2) Modify the `zed.launch` file setting the `resolution` to VGA, otherwise the YOLO node will crash due to not enough available memory: 
`<arg name="resolution"           default="3" />`
3) Open another terminal and launch the zed ROS wrapper: 
`$ roslaunch zed_wrapper zed.launch`
4) Change the name of the image topic used by YOLO in the file `ros.yaml` in the folder `config` of `darknet_ros`: 
```
    camera_reading:
    topic: /zed/rgb/image_rect_color
    queue_size: 1
```
5) Open another terminal and launch the YOLO node:
`$ roslaunch darknet_ros darknet_ros.launch`

Since this is not an issue related to ZED, please refer to the `darknet_ros` repository for other questions.

Myzhar, 2018-09-08:
@chijepark have you solved? Can we close this issue?

closed 2018-09-24 by Myzhar

JiyongOh, 2018-10-30:
Check the topic name of 'camera_reading' in the file of /darknet_ros/darknet_ros/config/ros.yaml.
I used '/camera/image_raw' instead of '/camera/rgb/image_raw'

virraaj, 2019-07-05:
Hi @Myzhar, Can you give some hint if I want to get distance to the object detected using ZED camera in darknet_ros,  what must be my steps?

Myzhar, 2019-07-05:
Hi @virraaj ,
can you open a new issue since this is a new question?

Thank you
Walter