stereolabs/zed-ros-wrapper #263: ZED stereo calibration using ROS

ravikt, 2018-08-24:
Hello,

Is there any way to overwrite the ZED calibration with parameters obtained from ROS stereo calibration ?
The SNXXX.conf file does not provide options for all the parameters to be edited. With focal length and principal points, the camera performs autocalibration. I want the ZED camera to use parameters obtained from ROS (D, K, R and P matrices). 

thank you

Ravi

Myzhar, 2018-08-24:
The SDK does not allow to use parameters calculated externally. If you want to perform your own stereo processing you can get both left and right raw images using the ROS wrapper or writing your ROS node that only retrieves the two images without performing depth calculation

ravikt, 2018-08-24:
How does the autocalibration work, in case I provide my own fx, fy, cx, cy, k1 and k2 values ?

Myzhar, 2018-08-24:
I'm sorry, but this is not public.

ravikt, 2018-08-24:
Is the ZED's visual odometry algorithm public ?

Myzhar, 2018-08-24:
No. The SDK is closed. Is public what you can find on Github

closed 2018-08-27 by Myzhar