stereolabs/zed-ros-wrapper #265: point_cloud in between .svo and .bag

kyumanlee84, 2018-08-27:
1. I recorded a *.svo file using ZED Explorer and replay it using ZED Depth Viewer to check if the recorded point_cloud is enough for my research regarding collision check. The point_cloud is dense and as fast as frame_rate:=30 resolution:=1080p. It looks good to use PCL. 
[image]

2. However, when I recorded point_cloud at same location using "**roslaunch zed_wrapper zed.launch resolution:=1 frame_rate:=30 camera_model:=0**" with "rosbag record -a", "rviz" shows that the recorded point_cloud is even more sparse (scatter) and most space is vacant or black and too slow. Also, I got the following errors:
[image] 
although I set "**rosparam set /zed/zed_wrapper_node/openni_depth_mode 1**".

Basically, with same ZED at same location, do generated point_cloud depend on which method (ZED tools or ROS) is used?

3. In addition, I tried to convert the *.svo to *.bag by running "r**oslaunch zed_wrapper zed.launch resolution:=1 frame_rate:=30 camera_model:=0 svo_file:=.../Documents/ZED/...svo**"  with "rosbag record -a", I got the same error above. At this case, the size of bag file is too large and the speed of bag file is too small..

Even though I ran "roslaunch zed_wrapper zed.launch resolution:=1 frame_rate:=30 camera_model:=0 svo_file:=.../Documents/ZED/...svo" and "rviz" to see recorded point_cloud without re-saving as .bag, this yields same error as follows:
[image]
[image]

[image]

In sum, which setting can provide the most dense and fastest point_cloud? 
Thanks,

kyumanlee84, 2018-08-29:
Thanks for the quick answers. I have further questions as follows:

1. To run an SVO file, is "**rosparam set /zed/zed_wrapper_node/openni_depth_mode 1**" required?
2. Although I set "**rosparam set /zed/zed_wrapper_node/quality 4**", I could not see any difference. Does this setting affect the number of the collected point cloud by SVO? Or for just visualization purpose in rivz?
3. For displaying as much as point cloud, is the value of the confidence set to 100 because of below?
"confidence threshold : confidence is a threshold filtering the values of the depth or the point cloud. With a confidence threshold set to 100, all the depth values will be written in the depth and the point cloud. You need to lower the threshold (80 by default) to remove the least accurate values."
My algorithm includes some filters for filtering out noise and non-necessary points. If the confidence is changed to lower the threshold, then does some filter works? If yes, which filter?
Also, although I play with the confidence value, I could not see any difference as well.

Thanks,

kyumanlee84, 2018-08-29:
One more question when recording an SVO file by **ZED mini.**
1. I recorded the SVO file using ZED Explorer and playback it at ZED Depth Viewer. Is there any tools or API to check if the SVO file include measured imu raw data? 
2. When I playback the SVO file using roslaunch zed_wrapper, "/zed/imu/data_raw" is on "rostopic list" but no values there, as shown in below.
[image]
It seems that imu data are not correctly saved. 
On the other hand, when I playback a ROS bag file recorded directly by roslaunch zed_wrapper, the /zed/imu/data_raw topic has values.

Here is another question regarding the setting and mode.
3. To compare, the default setting of ZED Depth Viewer is Ultra mode, 97% confidence, and 15m max range. Do these settings work for only visualization/display purpose when playing back? If I test a realtime experiment with ZED, do the changes of the settings affect the quality/density of (realtime) generated point cloud?

Thanks,

nesnes, 2018-08-30:
Hi,
When recording an SVO file with the ZED Mini, the raw IMU data aren't exported.
However, at each frame, a quaternion resulting from the accel+gyro fusion is exported to represent the camera orientation.
From the C++ SDK, you can get it using [getIMUData(..., TIME_REFERENCE_IMAGE)](https://www.stereolabs.com/developers/documentation/API/v2.5.1/classsl_1_1Camera.html#a453890423c830b627fc35c63edde7615).

If you do need the raw data, a ROS bag or a custom export might be a solution.

Myzhar, 2018-09-24:
@kyumanlee84  I close this issue, but you can continue to post comments if you need further help.

closed 2018-09-24 by Myzhar