stereolabs/zed-ros-wrapper #266: The rate of publish /zed/right/image_raw_color is so low!?

yueshukun, 2018-08-28:
@Myzhar 
Hi,
     I am running zed on notebook intel i7 NVIDIA GEFORCE GTX 960m ,and I set resolution=2, frame_rate=60.But the frame rate of /zed/right/image_raw_color which i am getting is around 4hz.How do I increase the frame rate? Thanks.

Myzhar, 2018-08-28:
I use a laptop with exactly the same configuration and I can get the maximum framerate:
```
$ rostopic hz /zed/right/image_raw_color
subscribed to [/zed/right/image_raw_color]
average rate: 60.281
	min: 0.014s max: 0.019s std dev: 0.00109s window: 56
average rate: 60.111
	min: 0.014s max: 0.020s std dev: 0.00128s window: 116
average rate: 60.086
	min: 0.013s max: 0.020s std dev: 0.00131s window: 176
average rate: 60.039
	min: 0.012s max: 0.025s std dev: 0.00147s window: 236
average rate: 60.038
	min: 0.012s max: 0.025s std dev: 0.00139s window: 296
average rate: 60.020
	min: 0.012s max: 0.025s std dev: 0.00142s window: 356
average rate: 60.025
	min: 0.012s max: 0.025s std dev: 0.00138s window: 416
average rate: 60.020
	min: 0.012s max: 0.025s std dev: 0.00137s window: 476
^Caverage rate: 60.018
	min: 0.012s max: 0.025s std dev: 0.00137s window: 494
```
How do you measure it?

yueshukun, 2018-08-28:
@Myzhar 
     I'm so confused, I use zed sdk 2.4.1, CUDA 9.1. Do you use them? or the lastest sdk? Do you change other parameters?

obraun-sl, 2018-08-28:
If you have 4Hz for 720p resolution, it's most probably because the camera is running in USB2.0 (if it's a ZED and not a ZED-M). Check that the camera is properly connected to an USB3.0 port.

yueshukun, 2018-08-28:
@obraun-sl You are right! I just run zed sdk smaple. The rate is so low too! Thanks.

closed 2018-08-28 by yueshukun