stereolabs/zed-ros-wrapper #267: Point Cloud difference between TX2 and Ubuntu PC

ArkadiuszNiemiec, 2018-09-05:
I have ZED-M working on TX2 with J120 carrier board but the Cloud Point is very bad in comparison to the same config but on Ubuntu PC. For example, PC' cloud clearly shows flat wall but TX2's cloud shoots way over that wall and does not resemble anything.

Do you have any tips on how to make ZED-M work on TX2? Do I have to perform a calibration to make it work? It downloaded a calibration file at first run but don't know if it is valid...

Myzhar, 2018-09-05:
If you think that it's a calibration problem you can take the calibration file from the laptop and copy it to TX2 (make a backup of the existing file to be sure).
If the result is not the same even using the same calibration files we can investigate deeper.
Please be sure that DEPTH_MODE is the same on both machines

ArkadiuszNiemiec, 2018-09-10:
@Myzhar Thank you for your answer. I double checked and both configs were the same but now the PC's point cloud is similar to my TX2... I am blaming lightning, for now, so I will close this issue and investigate further.

closed 2018-09-10 by ArkadiuszNiemiec

ArkadiuszNiemiec, 2018-12-11:
@nominator you have to install ZED SDK on both machines but I believe that they are identical. I have same CUDA and SDK versions on all devices and I was able to transfer my ROS projects without any problem.

nominator, 2018-12-11:
@ArkadiuszNiemiec thanks for getting back. So did you find out why point cloud was different on TX2?
and whether you got better quality point clouds on TX2. I haven't reached at the stage where I can compare. Once I do, I will share my results here.