stereolabs/zed-ros-wrapper #268: zed.resetTracking decrease the camera frame rate. from 30 to 24;

Garfield-kh, 2018-09-09:
In my code, I count the time when zed.pose_confidence > 40;
after the counter more than 20;
I ask the camera to do zed.resetTracking;
but I found that it freezes the camera and decrease the frame rate.
I have 2 zed_mini and 1 zed.  zed works okay but zed_mini cannot get frequency of 30.
I searched and found at https://github.com/stereolabs/zed-ros-wrapper/issues/161; but the solution update SDK 2.5.1 does not solve my problem.
Seek for help plz.
Thanks.

Myzhar, 2018-09-09:
Hi @Garfield-kh, 
can you post more information about the system on which you are having this problem?
How are you performing the "resetTracking"? Are you using an initial guess for the position or do you reset to the origin?

Myzhar, 2018-09-09:
Thank you for the detailed source code. So you are not using the official ROS node. 
I will test your node to try to replicate the problem

Can you tell me what kind of system are you working on? Laptop, desktop, NVidia Jetson...

Garfield-kh, 2018-09-09:
Thank you for your reply.
I am using TX2.

Garfield-kh, 2018-09-11:
I find that the confidence API for ZED and ZEDmini is different, and ZED mini we didn't find its confidence, which I think may cause the frame rate drop.

Garfield-kh, 2018-09-12:
Because confidence API for ZED and ZEDmini is different, and we cannot find the API for ZEDmini, the always to be 0 or -1 for ZEDmini, which makes the program resetTracking all the time and decrease the frame-rate.
I am going to email ZED company and ask for the API.

Garfield-kh, 2018-09-13:
I tried printing the confidence value on ZEDwrapper using Zedmini, it looks ok like 98 97, but in my personal code, it's not. What's wrong.. XD

Myzhar, 2018-09-13:
@Garfield-kh a first thing: when you post code on Github use markdown to format it (click "insert code" in the tool bar), so it's more clear.

Second, instead of using `COORDINATE_SYSTEM_RIGHT_HANDED_Z_UP` use the coordinate system made for ROS: `COORDINATE_SYSTEM_RIGHT_HANDED_Z_UP_X_FWD`.
Replace
`init_params.coordinate_system = COORDINATE_SYSTEM_RIGHT_HANDED_Z_UP ;`
with
`init_params.coordinate_system = COORDINATE_SYSTEM_RIGHT_HANDED_Z_UP_X_FWD ;`
so you must not perform coordinate changing manually if you get position in this way:
`TRACKING_STATE tracking_state = zed.getPosition(zed_pose, sl::REFERENCE_FRAME_WORLD);`
...
```
            _poseMsg.pose.position.x = zed_pose.getTranslation().tx;
            _poseMsg.pose.position.y = zed_pose.getTranslation().ty;
            _poseMsg.pose.position.z = zed_pose.getTranslation().tz;
            _poseMsg.pose.orientation.x = zed_pose.getOrientation().ox;
            _poseMsg.pose.orientation.y = zed_pose.getOrientation().oy;
            _poseMsg.pose.orientation.z = zed_pose.getOrientation().oz;
            _poseMsg.pose.orientation.w = zed_pose.getOrientation().ow;
```
this is available since SDK v2.5.1

The API for ZED and ZED mini is the same:
https://www.stereolabs.com/developers/documentation/API/v2.6.0/classsl_1_1Camera.html#a33271fce55b57b769c76e7b30a6d0f28

The new SDK v2.6.0 is available for download since yesterday:
https://www.stereolabs.com/developers/release/2.6/#sdkdownloads_anchor

Try with it and let me know if you continue to have different results.

Garfield-kh, 2018-09-13:
Thank you for your suggestion.
I update firmware and change the code COORDINATE_SYSTEM_RIGHT_HANDED_Z_UP;
but the problem is still not solved.

May I confirm with that API is the same for ZED and ZED mini for grab>getPosition?
I will try reading the wrapper code and see if it helps.
Thank you!

Myzhar, 2018-09-13:
Which SDK version are you using?

Is it the same if you change `sl::REFERENCE_FRAME_CAMERA` with `sl::REFERENCE_FRAME_WORLD`?

Garfield-kh, 2018-09-13:
the SDK now is 2.6.0;
Change to World frame is still not working.
There must be something wrong with my code. XD

Myzhar, 2018-09-13:
I'm investigating your code. Let me know if you find the solution before me.

Garfield-kh, 2018-09-13:
[test-zedTracking.tar.gz](https://github.com/stereolabs/zed-ros-wrapper/files/2378476/test-zedTracking.tar.gz)
this is the ROS file workspace, which has the problem. Could you help me have a try? 
Yes, of course, I will update you if I find any solution.
Thank you!

Myzhar, 2018-09-13:
I think that your problem is here:
`init_params.camera_disable_self_calib = false;`
Setting it to `true` I get correct values...

Why are you disabling self calibration?

Garfield-kh, 2018-09-13:
Oh!!!! It works!!

Myzhar, 2018-09-13:
I close the issue. You can comment anyway if you need further help

closed 2018-09-13 by Myzhar

Garfield-kh, 2018-09-14:
The reason why we disable self calibration is that: we mount ZED on drone application, because the drone keeps move, we are not sure if self_calib affects its VO performance. 
Thank you!