stereolabs/zed-ros-wrapper #269: What is the proper way to rename the tf_frames

sjhansen3, 2018-09-11:
Setting: the following argument when launching zed_camera.launch
    <arg name="right_camera_frame"   default="head_camera_frame"/>
does not result in a change to the tf tree frame names. This is confirmed by running view frames.

What is the recommended way to change the name of a zed tf frame?

sjhansen3, 2018-09-11:
It appears it might require changing the urdf - I string replaced all instances of the argument `right_camera_frame` and that seemed to do the trick.

sjhansen3, 2018-09-11:
But I am still curious if there is a more elegant way to do this or something I might be missing - It would be nice to not have to fork the zed-ros-wrapper simply to change the name of a frame. :)

Myzhar, 2018-09-11:
If you change a frame name in the launch file and the frame name is also used in the URDF, you must manually change it in the robot description file.
The alternative is to not use the URDF description and publish the pose of every joint in the code, even if the joint is static.

Myzhar, 2018-09-24:
@sjhansen3 I close this issue, but you can continue to post if you need further help

closed 2018-09-24 by Myzhar