stereolabs/zed-ros-wrapper #271: How to change Gain value in ROS?

vishnukumarts, 2018-09-13:
I referred your wiki page.  Under the "Dynamic Parameters" section there are few configuration data is there.  I am not sure how to use the data.  for example, if i want to increase the gain. what is the step i have follow.

Thanks,

Myzhar, 2018-09-13:
You can find the documentation for ROS Dynamic parameters here: http://wiki.ros.org/dynamic_reconfigure

You can change a dynamic parameter using the GUI provided by the `rqt` stack: 
`$ rosrun rqt_reconfigure rqt_reconfigure`
[image]

Remember that to be able to change the `gain` value you must be sure that `auto_exposure` is not enabled.

vishnukumarts, 2018-09-13:
Great. thank you.

closed 2018-09-13 by vishnukumarts