stereolabs/zed-ros-wrapper #273: /zed/odom vs /zed/map and wrong fixed frame

Haya-Alsharif, 2018-09-13:
Dear Team, 
I am using ZED Mini (ZED-M) camera and I would like to obtain the pose estimation by fusing the IMU and visual odometry data. 

According to the [ZED ROS wrapper documentation](https://docs.stereolabs.com/integrations/ros/zed_node/) this is already implemented and can be accessed easily through the topic `/zed/map` instead of `/zed/odom` where the latter does pure visual odometry without IMU integration.

My problem is that the values from `/zed/map` do not change as my camera moves around when the fixed frame is set to "map." `/zed/odom` does!  I thought that since they are both estimating the pose relative to the initial point, they both should have similar readings until one of them drift. The only way to get `/zed/map` to change with camera movement is when the fixed frame is set to "imu_link," and at that point, the `/zed/odom` topic stops moving.

**My questions are**
1. Can you please confirm that the **documentation** on the `/zed/odom` and `/zed/map` topic in your documentation page is correct?
2. Is it normal to get almost **fixed readings** for `/zed/map` topic when the fixed frame is set to "map"? Will it ever move and if yes, when and why?
3. Why the `/zed/map` vector moves only when the fixed frame is set to anything other than map, seems like a tf problem?
4. How can I use the `/zed/map` as an odometry topic in the RTABmap integration instead of `/zed/odom`?

Thank you very much

Myzhar, 2018-09-13:
Dear @Haya-Alsharif,
what you are describing is really weird. 
'/zed/map' should change moving the camera and you should see it using Rviz, mainly because '/zed/odom' is changing.

1. The documentation is correct 
2. No, as I told above
3. That's weird
4. You can't. RTABmap wants an odometry message and the pose is not an odometry. Odometry should be continuous, the pose is not. The tracking algorithm generates jumps and they are not allowed by the definition of odometry.
RTABmap calculates a pose starting from an odometry, using visual and spatial information, if you start from a pose you will get bad results.
You can read more about this in the REP105:
http://www.ros.org/reps/rep-0105.html

About the problem of the map pose not changing I need more information:
1) which version of the SDK are you using?
2) which version of the wrapper are you using?
3) how are you running the node? Launch file?
4) if you are using a launch file have you changed any parameter?

Before the end of the month a new version of the wrapper will be released. It will have some fixes and new features to follow the novelties introduced with the SDK v2.6, many just about the Pose Tracking

sarahToonsi, 2018-09-13:
@Myzhar
We need the fused position published as odometry, is that possible?! 
If not please explain why?!
If yes, how can we contact the developers to address this issue

Myzhar, 2018-09-13:
If you want to use the fused pose as an odometry you must modify the code of the wrapper changing the type of the topic published by 'purpose here:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L470.
The topic must be changed from '
geometry_msgs::PoseStamped' to 'nav_msgs::Odometry'

Then you must modify the function 'publishPose' 
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L832-L849
to fill the field of the new message type, you can follow the function 'publishOdom' as example

https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L820-L830

but remember that the fused position generated by a SLAM algorithm is not an odometry and you can have problems if you use it in that way.

sarahToonsi, 2018-09-13:
@Myzhar 
Im sorry but I didnt get why a fused position cant be an odometry. As far as I know, sensor fusion enhances the vision abilities to calculate odometry, which means that the sensor fusion shouldnt only give odometry as the vision one, but also enhanced and more accurate.
Its like you are doing the same vision job but adding IMU to fix drifts and errors thats all.

Myzhar, 2018-09-13:
The situation is well described in the REP105 that I linked above.
As I told before the odometry pose must be continue, instead the pose generated by the position tracking algorithm (mainly if 'spatial_memory' is avtive) can have jumps and they "philosophically" are not allowed for odometry. 

When you fuse Visual Odometry with Inertial Data from an IMU the final pose can be used as "enhanced" odometry because it is still continuous.

It is like when you use the GPS data to fix the position in outdoor applications, the resulting pose is not an odometry because the GPS data generate jumps in the position every second.

adujardin, 2018-09-14:
Hi,
The confusion come from the documentation. It is not correct. When using the ZED-Mini :
1. `zed/odom` is actually the visual-inertial information (sensor-fusion)
2. `zed/map` is visual-internal information + the SLAM module that can close loop or relocalize the camera (and therefore jump)

You can't access visual only tracking information when using a ZED-Mini.

We have to clarify/fix the documentation.

Haya-Alsharif, 2018-09-14:
@adujardin Thank you very much and I am very glad to hear that! 

Concerning `zed/map` topic, does that mean your camera comes with a SLAM module that performs the pose graph optimization and producing the pose and map information? If that is true, it is such a great feature that probably no one knows about. We are currently using the g2o LM optimizer through RTABmap to do that, but it will be much more convenient to use the generated pose and map by your SLAM module.

Btw, how can we access the **optimized map point cloud**? and can you confirm that the `zed/map` topic will print the **3D position information only** (the orientation barley changes) relative to the built map?

adujardin, 2018-09-14:
The ZED SDK includes a SLAM module with trajectory optimization yes. However, you won't be able to have exactly the same thing as RTABMap since you can't have access to previous positions to correct the point clouds positions.
For instance, if a loop closure is detected and the trajectory is optimized, with the ZED SDK you will only have the current (and future) position corrected, not the full correction along the graph. Therefore the previous point clouds integrated into the map will stay potentially misplaced, only the future one will have the corrected position.

TLDR: The ZED SDK has a SLAM module but you won't be able to get the fully optimized map point cloud like you do from RTABMap.

Haya-Alsharif, 2018-09-14:
Thank you @adujardin and @Myzhar. That solves all my issues.

closed 2018-09-14 by Haya-Alsharif