stereolabs/zed-ros-wrapper #280: Publishing ZED camera image through ROS without CUDA dependency?

sebo313, 2018-09-20:
Hello,
i would like to publish the right or left image of ZED camera only for using it somewhere else. Do i have to install all ZED SDK including CUDA dependency or is there an easier way of getting the image?

Myzhar, 2018-09-20:
You must install ZED SDK (that requires CUDA) only if you want to use the SDK capabilities.

If not, the ZED camera stream can be acquired as a standard UVC compatible camera. In this case you will receive an unique frame with the right and left unrectified image side by side, then you need to crop them,apply a rectification if you need it and send them as "normal" images.

sebo313, 2018-09-20:
My problem refers to
https://github.com/intel/ros_intel_movidius_ncs/blob/master/doc/video_detection.md#mobilenet_ssd

So when using "Option 2: USB camera " streaming nodelet and having the ZED camera plugged in as standard UVC  camera, there is no proper video image. It looks like this:

[image]

With a standard USB webcam i get a clear video image and it works perfectly, but i need to get run with ZED camera.
So my idea is to replace "Option 2: USB camera " streaming nodelet by a ZED camera left image streaming nodelet. Any idea how to get there?

Myzhar, 2018-09-20:
It seems a format problem. Remember that the ZED frame format is RGBA

Are you using OpenCV?

Myzhar, 2018-09-20:
You must work on these parameters:

https://github.com/intel/ros_intel_movidius_ncs/blob/devel/movidius_ncs_launch/launch/ncs_camera.launch#L53-L67

Myzhar, 2018-09-20:
I think that's the visualization callback, it takes the frame and the synchronized bounding boxes of the detected objects.

You must search for the callback that takes the image and starts the classification... one of that can be this:
https://github.com/intel/ros_intel_movidius_ncs/blob/26b8e2b6a04503befd3f260b8d06886087bd15de/movidius_ncs_stream/src/ncs_nodelet.cpp#L224

But maybe that it is not the only one

sebo313, 2018-09-20:
Yes thats the callback i am adressing: https://github.com/intel/ros_intel_movidius_ncs/blob/26b8e2b6a04503befd3f260b8d06886087bd15de/movidius_ncs_stream/src/ncs_nodelet.cpp#L248-L249

But how can i crop the image in there? Sorry for my bad cpp skills

Myzhar, 2018-09-20:
Yes, you must crop the "camera_data"

Something like:

```
cv::Rect roi;
roi.x = 0;
roi.y = 0;
roi.width = 1280;
roi.height = 720;

cv::Mat cropped = image_data(roi);

FUNP_D detection_result_callback = boost::bind(&NCSImpl::cbGetDetectionResult, this, _1, _2);
  ncs_manager_handle_->detectStream(cropped, detection_result_callback, image_msg);
```

sebo313, 2018-09-20:
Thanks so much!! Your code looks reasonable and does not throw an error but when running it it does not crop the image :( also tried to set roi.width to any number and it does not react..

sebo313, 2018-09-20:
i could fix it with pasting your code in here: https://github.com/intel/ros_intel_movidius_ncs/blob/26b8e2b6a04503befd3f260b8d06886087bd15de/movidius_ncs_example/src/stream_detection.cpp#L84

Thanks so much! You help a lot!!

closed 2018-09-20 by sebo313

agrija9, 2021-07-13:
I also faced the corrupted image with the ZED camera. 

I didn't use OpenCV directly, rather I managed to run the camera using the usb_cam ros driver (https://github.com/ros-drivers/usb_cam).

After installing the driver, I modified the launch file to:

`
<launch>
  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video2" />
    <param name="image_width" value="2560" />
    <param name="image_height" value="720" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="framerate" value="30" />
    <param name="io_method" value="mmap"/>
  </node>

  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>
`

After building the node, you can launch it via `roslaunch usb_cam usb_cam-test.launch`, this will visualize the stereo output and publish to a specific image topic.