stereolabs/zed-ros-wrapper #281: Add an ability to turn of /map -> /odom transform to use with RTABMAP

ArkadiuszNiemiec, 2018-09-21:
I have found an issue that makes ZED-M not able to work with RTABMAP while using ZED's odometry and RTABMAP loop-closure. My goal is to use odometry from ZED and allow RTABMAP to publish `/map -> /odom` and correct it when loop-closure is detected.
The problem is that `zed-ros-wrapper` already publishes `/map -> /odom` transformation and thus both software overwrites themselves.

To solve it I commented a few lines in `zed-ros-wrapper` to stop it from publishing this transform. Do you think that we can add a parameter that turns it off? Do you want me to do a PR?

Myzhar, 2018-09-21:
If you are using the latest updated version from the `master` branch you can find an example about how to correctly set the wrapper to be used with RTABmap:
https://github.com/stereolabs/zed-ros-wrapper/tree/master/zed_rtabmap_example

Are you using it?

ArkadiuszNiemiec, 2018-09-21:
@Myzhar Yes but this example is using odometry from RTAB-Map. My goal is to use odometry from ZED and still allow RTAB-Map to correct it on loop-closure.

closed 2018-09-21 by ArkadiuszNiemiec

Myzhar, 2018-09-21:
Well, so the parameter I cited in the previous comment is what you are searching for.
Try to checkout the `tutorial_feat` branch and to set `publish_map_tf` to `false` and let me know if that's what you were searching for.
If you have updated to latest SDK v2.6 you will also get the odometry covariance.

ArkadiuszNiemiec, 2018-09-21:
@Myzhar I have just tested it. That's exactly what I needed. Thank you.

ArkadiuszNiemiec, 2018-09-21:
@Myzhar Unfortunately RTAB-Map crashes with this repo:
```[FATAL] (2018-09-21 10:28:55.216) OptimizerG2O.cpp:533::optimize() Condition (optimizer.verifyInformationMatrices()) not me```

I have found this topic: [RTAB-Map forum](http://official-rtab-map-forum.67519.x6.nabble.com/g2o-error-on-loop-closure-td4707.html#a4725) in which the author says:

> The error is caused by bad covariance set in odometry messages. 

Is there a way to turn off covariance? I am tryting to debug it furher.

Myzhar, 2018-09-21:
Since there is not a loop closure the covariance will always increase if the odom is never reset.
I'm going to add a parameter to disable covariance publishing.
It will be ready in a few minutes

Myzhar, 2018-09-21:
@ArkadiuszNiemiec if you get the latest  update from the `tutorials_feat` branch you can find the parameter `publish_pose_covariance` in the file `zed_camera.launch`

Myzhar, 2018-09-21:
You're welcome. Thank you too. User feedback are the best way to understand what's missing or wrong 👍