stereolabs/zed-ros-wrapper #284: Unable to catkin_make the zed-ros-wrapper-2.2.x

koshish, 2018-09-23:
Dear all,

I am unable to catkin_make the the zed-ros-wrapper-2.2.x package in my system.

ros: kinetic
package: zed-ros-wrapper-2.2.x 
OS: Ubuntu 16.04

$ cat /usr/local/cuda/version.txt
CUDA Version 8.0.61

According to https://www.stereolabs.com/developers/release/2.2/  the package should work with CUDA version 8 . But, it failed to catkin_make the package. I have included the out put of the commands below. 

$ nvidia-smi
Sun Sep 23 23:17:33 2018       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.87                 Driver Version: 390.87                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Quadro 2000         Off  | 00000000:03:00.0  On |                  N/A |
| 30%   42C   P12    N/A /  N/A |    261MiB /   959MiB |      1%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1079      G   /usr/lib/xorg/Xorg                           183MiB |
|    0      1921      G   compiz                                        73MiB |
|    0     13354      G   /usr/lib/firefox/firefox                       1MiB |
+-----------------------------------------------------------------------------+

hercules-16-04@hercules-16-04-Precision-T3600:~/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws$ catkin_make
Base path: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws
Source space: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/src
Build space: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build
Devel space: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/devel
Install space: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed-ros-wrapper-2.2.x)
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CUDA: Found unsuitable version "8.0", but required is exact
  version "9.2" (found /usr/local/cuda)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_args)
  /usr/local/zed_previous/zed-config.cmake:42 (find_package)
  zed-ros-wrapper-2.2.x/CMakeLists.txt:22 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:318: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

This is the output of diveceQuery from the CUDA 8.0 samples
hercules-16-04@hercules-16-04-Precision-T3600:~/NVIDIA_CUDA-8.0_Samples/1_Utilities/deviceQuery$ ./deviceQuery 
./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

Detected 1 CUDA Capable device(s)

Device 0: "Quadro 2000"
  CUDA Driver Version / Runtime Version          9.1 / 8.0
  CUDA Capability Major/Minor version number:    2.1
  Total amount of global memory:                 960 MBytes (1006108672 bytes)
  ( 4) Multiprocessors, ( 48) CUDA Cores/MP:     192 CUDA Cores
  GPU Max Clock rate:                            1251 MHz (1.25 GHz)
  Memory Clock rate:                             1304 Mhz
  Memory Bus Width:                              128-bit
  L2 Cache Size:                                 262144 bytes
  Maximum Texture Dimension Size (x,y,z)         1D=(65536), 2D=(65536, 65535), 3D=(2048, 2048, 2048)
  Maximum Layered 1D Texture Size, (num) layers  1D=(16384), 2048 layers
  Maximum Layered 2D Texture Size, (num) layers  2D=(16384, 16384), 2048 layers
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 32768
  Warp size:                                     32
  Maximum number of threads per multiprocessor:  1536
  Maximum number of threads per block:           1024
  Max dimension size of a thread block (x,y,z): (1024, 1024, 64)
  Max dimension size of a grid size    (x,y,z): (65535, 65535, 65535)
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and kernel execution:          Yes with 1 copy engine(s)
  Run time limit on kernels:                     Yes
  Integrated GPU sharing Host Memory:            No
  Support host page-locked memory mapping:       Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support:                        Disabled
  Device supports Unified Addressing (UVA):      Yes
  Device PCI Domain ID / Bus ID / location ID:   0 / 3 / 0
  Compute Mode:
     < Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) >

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version = 9.1, CUDA Runtime Version = 8.0, NumDevs = 1, Device0 = Quadro 2000
Result = PASS

Am, I missing something during the setup process ? Any suggestions?

Regards,
Koshish Koirala
American University of Sharjah, UAE

Myzhar, 2018-09-23:
Which version of the ZED SDK have you installed?
According to the error you pasted the problem is with the file "CMakeLists.txt" that searches for CUDA 9.2.
Search for the line 
`find_package(CUDA 9.2)`
and try to modify it in 
`find_package(CUDA)`
If the line is just in the second version then it's the SDK that is searching for CUDA 9.2.

Let me know

Myzhar, 2018-09-23:
Ok, I thought that your problem was this but I wanted to be sure:
When you install a version of a library, you compile a ROS package with it, then you change the version of that library, you must clear the cache of CMake to let it find the new library, otherwise it will continue to search for the former version.
So try to delete the "devel" and the "install" folder (not the "src" 😉) and try to compile again

koshish, 2018-09-23:
> So try to delete the "devel" and the "install" folder (not the "src" wink) and try to compile again

I installed a old version of library and download the old version of package and tried to compile.

I tried your above suggestion to delete the directorys "devel" and "build"  and compiled. Still it is looking for Cuda version 9.2 .  :(

`zed_v_2_2_x_catkin_ws$ ls`
**build  devel  src**
`zed_v_2_2_x_catkin_ws$ rm -rf devel/`
`zed_v_2_2_x_catkin_ws$ rm -rf build/`
`zed_v_2_2_x_catkin_ws$ catkin_make`

-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_wrapper'
-- ==> add_subdirectory(zed-ros-wrapper-2.2.x)
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CUDA: Found unsuitable version "8.0", but required is exact
  version "9.2" (found /usr/local/cuda)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_args)
  /usr/local/zed_previous/zed-config.cmake:42 (find_package)
  zed-ros-wrapper-2.2.x/CMakeLists.txt:22 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/hercules-16-04/ros-works/ros-workspaces/zed_v_2_2_x_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Myzhar, 2018-09-23:
Ok. Maybe I found your problem in this line:
 `/usr/local/zed_previous/zed-config.cmake:42 (find_package)`
When you installed the SDK v2.2 you chose to keep the latest version as backup in `/usr/local/zed_previous`.
For a weird reason Ubuntu continues to use it instead of the newer.
Try to delete that folder

koshish, 2018-09-23:
> `/usr/local/zed_previous/zed-config.cmake:42 (find_package)`
> When you installed the SDK v2.2 you chose to keep the latest version as backup in `/usr/local/zed_previous`.
> Try to delete that folder

Finally it worked. `/usr/local$ sudo rm -rf zed_previous/` :smile: 

Thank You so much. You have amazing debugging skills :wink: 

>For a weird reason Ubuntu continues to use it instead of the newer.
Isn't it supposed to dlink the old installation folder after installing the new SDK library? I wouldn't have kept that folder if I knew it would still search for the old folder.

Just out of curiosity is there a workaround to link the new zed folder without deleting the zed_previous folder?

Myzhar, 2018-09-23:
Maybe that a solution has been found for the newer SDK. I must ask for it

closed 2018-11-24 by Myzhar