stereolabs/zed-ros-wrapper #286: Cmake Issue + missing imu_link tf frame + unity as transform frame between links.

rhklite, 2018-09-24:
Hi,

I'm fairly new to ROS, please excuse me if my questions seems ignorant.

I'm currently have a ZED mini, station: Nvidia Quadro K2200, ZED SDK 2.6.0 (CUDA 9.0 version) Ubuntu 16.04, ROS Kinetic

I've encountered 2 issues, i'm not sure if these are related.

1. Cmake issue
this error shows up during catkin build, though i already did sudo apt install libomp-dev
```
CMake Error at /home/richard/catkin_ws/src/zed-ros-wrapper/zed_wrapper/CMakeLists.txt:17 (message):
OpenMP not found, please install it to improve performances: 'sudo apt install libomp-dev'
Call Stack (most recent call first):
CMakeLists.txt:34 (checkPackage)

```
my temporary solution was to comment out the find_package(OpenMP) section of the code  in zed_wrapper/CMakeLists.txt, and set the following flags  to attemp to still enable OpenMP
```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} ${OpenMP_C_FLAGS} -fopenmp")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS} -fopenmp")
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} ${OpenMP_EXE_LINKER_FLAGS} -fopenmp")
```

This compiled fine, and the example demos all seems ok, but I'm not sure if this has an affect on anything else.

2. missing imu_link transform in rqt_tf_tree, and not found in tf_view
I launched this file `roslaunch zed_wrapper zed.launch` But could not see the imu_link. I checked the zedm.urdf file, and i do see this section of the code
```
        <joint name="imu_link_joint" type="floating">
            <parent link="zed_camera_center"/>
            <child link="imu_link"/>
            <origin xyz="0 0 0" rpy="0 0 0" />
        </joint>

        <link name="imu_link" />
```
I'm not sure why this isn't getting published.

when i do rosrun tf tf_echo /someframe /imu_link
i get 
`Exception thrown:"imu_link" passed to lookupTransform argument source_frame does not exist. `

Myzhar, 2018-09-24:
Hi @rhklite

1. Are you sure that OpenMP is available in your system? Try installing it:
`sudo apt install libomp-dev`

2. The IMU frame `imu_link` is published only if there is at least a subscriber for the topic `/zed/imu/data`
Try to open a new console and try to subsdrive to it using
`rostopic echo /zed/imu/data` 
then try to use `tf_echo` again

Myzhar, 2018-09-24:
Can you verify that the runtime libraries are installed?
`$ sudo apt install libomp5`

rhklite, 2018-09-24:
hi, the libomp5 is installed
`sudo apt install libomp5` output: `libomp5 is already the newest version (3.7.0-3).`
`cmake --version` output: `cmake version 3.9.0`

Myzhar, 2018-09-24:
It seems an issue due to the version of CMake that you are using, I can not replicate the issue on Ubuntu 16.04 with CMake v3.5.1

rhklite, 2018-09-24:
Thank you. Do you know if and how the work around i initially stated might affect the performance of the wrapper?

Myzhar, 2018-09-24:
I don't think it can affect the performance of the wrapper. You manually set the configuration to use OpenMP, so you are using it.

closed 2018-09-24 by rhklite

hanzheteng, 2020-08-20:
> 2. The IMU frame `imu_link` is published only if there is at least a subscriber for the topic `/zed/imu/data`
>    Try to open a new console and try to subscribe to it using
>    `rostopic echo /zed/imu/data`
>    then try to use `tf_echo` again

It would be great if this information can be added to [the documentation](https://www.stereolabs.com/docs/ros/zed-node/#transform-frames). I spent quite a while finding why imu_link is not available.