stereolabs/zed-ros-wrapper #289: Calibration parameters to use with ORB_SLAM2?

ArkadiuszNiemiec, 2018-09-25:
I am trying to test ZED-M with ORB_SLAM2 RGB-D. If I understand it correctly ZED publishes already rectified images so I only need to edit the first section of this [https://github.com/raulmur/ORB_SLAM2/blob/master/Examples/Stereo/KITTI04-12.yaml](example file).

Could you explain how to get them?

```# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 707.0912
Camera.fy: 707.0912
Camera.cx: 601.8873
Camera.cy: 183.1104

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0

Camera.width: 1241
Camera.height: 376

# Camera frames per second 
Camera.fps: 10.0

# stereo baseline times fx
Camera.bf: 379.8145

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40
```

I thought that `/zed/rgb/camera_info` topic might be useful but it's empty for me.

Myzhar, 2018-09-25:
The `/zed/rgb/camera_info` topic is published only if the relative image is subscribed, so you must open a console and subscribe to `/zed/rgb/image_rect_color`:
Console 1:
`$ rostopic hz /zed/rgb/image_rect_color`
Console 2:
`$ rostopic echo /zed/rgb/camera_info`

ArkadiuszNiemiec, 2018-09-25:
Thanks, I was able to run ORB_SLAM2 but it's working good enough in RGBD mode... My hunt for a loop-closure software continues :smile:

closed 2018-09-25 by ArkadiuszNiemiec

rhklite, 2018-10-04:
@ArkadiuszNiemiec I'm also trying to use Zed mini with a SLAM package aswell. Have you tried using [maplab](https://github.com/ethz-asl/maplab/wiki) with it? If so, did you manage to get it to work?