stereolabs/zed-ros-wrapper #290: Using zed in ros as fancy webcam

sjhansen3, 2018-09-25:
Is there a way to use zed with ros as a fancy webcam? Some users (who havent installed cuda) of my setup will want to have the minimum functionality of using the zed's webcam. Is there a flag that can be set in order to allow this?

Ideally IMHO the behavior should be that a warning is thrown if CUDA is not detected, the depth detection does not work and the color topics still broadcast. Is this a thing?

closed 2019-01-17 by Myzhar

AndreV84, 2024-09-29:
@Myzhar is it possible to get stereolabs zed camera to work on windows as monocular camera ?

Myzhar, 2024-09-30:
The ZED USB cameras send out a not-rectified synchronized side-by-side stereo image stream using UVC standard.
You can only use them as "monocular" cameras if you perform a little processing to split the frames into two separate images.