stereolabs/zed-ros-wrapper #292: grab() frequency at 1fps

oneway234, 2018-09-26:
I am trying to use zed-ros-wrapper to  record images at 1FPS. Those options in zed-ros-wrapper are too high for me. How should I edit the ROS wrapper to reduce the grab() frequency at 1fps?

Myzhar, 2018-09-26:
Yes, the ZED wrapper is designed to grab data matching the camera framerate.
If you want to grab using a frequency slower than the camera framerate you must modify the code of the wrapper:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1469

Replace `mCamFrameRate` with the frequency you want to use.

oneway234, 2018-10-04:
I tried to replace mCamFrameRate to 1. But failed. It seems still grab images at 30fps.
I used ZED am I also need to modify the code of the wrapper?:
https://github.com/stereolabs/zed-ros-wrapper/blob/e736203c943ab0187cbb09ce371802dd0559b9b1/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1469

Myzhar, 2018-10-05:
I can confirm that my proposed solution works as expected:
```
    void ZEDWrapperNodelet::device_poll_thread_func() {
        //ros::Rate loop_rate(mCamFrameRate);
        ros::Rate loop_rate(1);
```

Recompile the node and test it:
[image]

as you can see the average rate of the message is about 1 Hz.

Please notice that you will start to received warning messages of this kind:
> ZED (Init) >> WARNING: FPS is too low to enable positional tracking and spatial mapping. Consider using PERFORMANCE parameters

you can ignore it for your scope.

oneway234, 2018-10-05:
I just tried. And it worked. Thanks.

BTW I the line should be change is:
"https://github.com/stereolabs/zed-ros-wrapper/blob/e736203c943ab0187cbb09ce371802dd0559b9b1/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1481".

closed 2018-10-05 by oneway234

Myzhar, 2018-10-09:
@vincentrou the parameter exists, but allows you to choose only a frequency compatible with the resolution of the image.

If you need data at lower FPS I suggest you to set the `frame_rate` parameter in the launch file to the lowest value (15 FPS), then you can use the `throttle` command from the [topic_tools](http://wiki.ros.org/topic_tools/throttle) package to limit the frequency of the topic.
In this way you are sure that the ZED SDK works as expected using the least resources as possible.

For example:
`$ rosrun topic_tools throttle messages /zed/rgb/image_rect_color 1 /zed/rgb/image_rect_color_1_hz`

you can check that it works as expected with:
```
$ rostopic hz /zed/rgb/image_rect_color_1_hz 
subscribed to [/zed/rgb/image_rect_color_1_hz]
no new messages
average rate: 0.966
	min: 1.035s max: 1.035s std dev: 0.00000s window: 2
average rate: 0.970
	min: 1.027s max: 1.035s std dev: 0.00429s window: 3
no new messages
average rate: 0.979
	min: 1.003s max: 1.035s std dev: 0.01371s window: 4
average rate: 0.976
	min: 1.003s max: 1.035s std dev: 0.01278s window: 5
average rate: 0.980
	min: 1.002s max: 1.035s std dev: 0.01457s window: 6
average rate: 0.978
	min: 1.002s max: 1.035s std dev: 0.01430s window: 7
average rate: 0.977
	min: 1.002s max: 1.035s std dev: 0.01361s window: 8
average rate: 0.980
	min: 1.001s max: 1.035s std dev: 0.01483s window: 9
```

Myzhar, 2018-10-11:
As I told you it is a param that would make the ZED work with unwanted behaviors, so we prefer to not add it.

The good of ROS is that you can assemble different pieces when one of it misses what you require.