stereolabs/zed-ros-wrapper #293: No output on the odom topic

pvishal-carnd, 2018-09-26:
Hi,

I am using a ZED-M camera with this wrapper on ROS and I have intermittent issues with the odometry not being published. I am using the latest 02a43a6 version. I run the unmodified `zed_wrapper/zed.launch` file. 

The node starts out well with the `/zed/odom` topic being advertised but the topic itself is not published. I have a screencast of this here: https://asciinema.org/a/metdgJe8DqMLE3oIkvc6M9BGz 

The ZED_Positional_Tracking example code works well for me all the time.

The results from your ZED_Diagnostics tool are [here](https://github.com/stereolabs/zed-ros-wrapper/files/2420661/ZED_Diagnostic_Results.txt). Let me know if you know any more information or if there is something I am missing.

Thanks!
Vishal

Myzhar, 2018-09-26:
The new wrapper has `floor_alognment` enabled as default. The odometry is not published until the floor is detected.
You can try two different things:
1. Move the camera versus the floor to detect it and start the positional tracking process
2. Disable `floor_alignment` in the `zed_camera.launch` 

The new documentation with the latest added parameters is coming in the next days.

pvishal-carnd, 2018-09-27:
Thanks! That was the issue indeed. The workspace this is working in is quite cluttered and disabling `floor_alignment` worked for me.

closed 2018-09-27 by pvishal-carnd