stereolabs/zed-ros-wrapper #298: still has a same bug about catkin_make in the zed-ros-wrapper

guan4286, 2018-09-28:
After super .I do it again .so i meet the same problem

root@g-System-Product-Name:/home/g/catkin# catkin_make
Base path: /home/g/catkin
Source space: /home/g/catkin/src
Build space: /home/g/catkin/build
Devel space: /home/g/catkin/devel
Install space: /home/g/catkin/install
####
#### Running command: "make cmake_check_build_system" in "/home/g/catkin/build"
####
####
#### Running command: "make -j8 -l8" in "/home/g/catkin/build"
####
[  0%] Built target geometry_msgs_generate_messages_cpp
[  3%] Built target fast
[  3%] Built target sensor_msgs_generate_messages_cpp
[  9%] Built target viso2
[  9%] Built target sensor_msgs_generate_messages_lisp
[  9%] Built target sensor_msgs_generate_messages_py
[  9%] Built target geometry_msgs_generate_messages_py
[  9%] Built target _svo_msgs_generate_messages_check_deps_Feature
[  9%] Built target _svo_msgs_generate_messages_check_deps_DenseInput
[  9%] Built target geometry_msgs_generate_messages_lisp
[  9%] Built target _svo_msgs_generate_messages_check_deps_NbvTrajectory
[  9%] Built target _svo_msgs_generate_messages_check_deps_Info
[  9%] Built target geometry_msgs_generate_messages_nodejs
[  9%] Built target geometry_msgs_generate_messages_eus
[  9%] Built target sensor_msgs_generate_messages_eus
[  9%] Built target sensor_msgs_generate_messages_nodejs
[ 11%] Built target talker
[ 12%] Built target turtle_tf_listener
[ 13%] Built target listener
[ 16%] Built target frame_tf_broadcaster
[ 16%] Built target pcl_ros_gencfg
[ 17%] Built target turtle_tf_broadcaster
[ 17%] Built target std_msgs_generate_messages_eus
[ 17%] Built target std_msgs_generate_messages_cpp
[ 17%] Built target std_msgs_generate_messages_lisp
[ 17%] Built target std_msgs_generate_messages_py
[ 17%] Built target std_msgs_generate_messages_nodejs
[ 17%] Built target roscpp_generate_messages_lisp
[ 17%] Built target roscpp_generate_messages_nodejs
[ 17%] Built target roscpp_generate_messages_eus
[ 17%] Built target rosgraph_msgs_generate_messages_eus
[ 17%] Built target roscpp_generate_messages_cpp
[ 17%] Built target rosgraph_msgs_generate_messages_lisp
[ 17%] Built target rosgraph_msgs_generate_messages_cpp
[ 17%] Built target rosgraph_msgs_generate_messages_nodejs
[ 17%] Built target roscpp_generate_messages_py
[ 17%] Built target rosgraph_msgs_generate_messages_py
[ 24%] Built target vikit_common
[ 24%] Built target std_srvs_generate_messages_py
[ 24%] Built target actionlib_generate_messages_py
[ 24%] Built target std_srvs_generate_messages_nodejs
[ 24%] Built target actionlib_generate_messages_nodejs
[ 24%] Built target std_srvs_generate_messages_lisp
[ 24%] Built target tf_generate_messages_py
[ 24%] Built target actionlib_generate_messages_lisp
[ 24%] Built target actionlib_generate_messages_eus
[ 24%] Built target actionlib_msgs_generate_messages_lisp
[ 24%] Built target tf_generate_messages_cpp
[ 24%] Built target tf2_msgs_generate_messages_cpp
[ 24%] Built target tf_generate_messages_eus
[ 24%] Built target std_srvs_generate_messages_cpp
[ 24%] Built target actionlib_msgs_generate_messages_cpp
[ 24%] Built target actionlib_generate_messages_cpp
[ 24%] Built target tf_generate_messages_nodejs
[ 24%] Built target actionlib_msgs_generate_messages_eus
[ 24%] Built target actionlib_msgs_generate_messages_py
[... 94 lines cut ...]
                                                                      ^
/opt/ros/kinetic/include/ros/console.h:356:64: note: in definition of macro ‘ROSCONSOLE_PRINT_STREAM_AT_LOCATION_WITH_FILTER’
     __rosconsole_print_stream_at_location_with_filter__ss__ << args; \
                                                                ^
/opt/ros/kinetic/include/ros/console.h:398:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_STREAM_AT_LOCATION’
       ROSCONSOLE_PRINT_STREAM_AT_LOCATION(args); \
       ^
/opt/ros/kinetic/include/ros/console.h:568:43: note: in expansion of macro ‘ROS_LOG_STREAM_COND’
 #define ROS_LOG_STREAM(level, name, args) ROS_LOG_STREAM_COND(true, level, name, args)
                                           ^
/opt/ros/kinetic/include/rosconsole/macros_generated.h:113:43: note: in expansion of macro ‘ROS_LOG_STREAM’
 #define ROS_INFO_STREAM_NAMED(name, args) ROS_LOG_STREAM(::ros::console::levels::Info, std::string(ROSCONSOLE_NAME_PREFIX) + "." + name, args)
                                           ^
/opt/ros/kinetic/include/nodelet/nodelet.h:60:34: note: in expansion of macro ‘ROS_INFO_STREAM_NAMED’
 #define NODELET_INFO_STREAM(...) ROS_INFO_STREAM_NAMED(getName(), __VA_ARGS__)
                                  ^
/home/g/catkin/src/zed-ros-wrapper/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp:792:13: note: in expansion of macro ‘NODELET_INFO_STREAM’
             NODELET_INFO_STREAM("Floor Alignment : " << (trackParams.set_floor_as_origin ? "ENABLED" : "DISABLED"));
             ^
zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/build.make:86: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o' failed
make[2]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/src/nodelet/src/zed_wrapper_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:6458: recipe for target 'zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all' failed
make[1]: *** [zed-ros-wrapper/zed_wrapper/CMakeFiles/ZEDWrapper.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

guan4286, 2018-09-28:
this is the rectify code 

 27 #include <rosgraph_msgs/Clock.h>
    28 #include <nav_msgs/Odometry.h>
    29 #include <nav_msgs/Path.h>
    30 #include <sensor_msgs/Imu.h>
    31 #include <sensor_msgs/distortion_models.h>
    32 #include <sensor_msgs/image_encodings.h>
    33 #include <sensor_msgs/point_cloud2_iterator.h>
    34 #include <stereo_msgs/DisparityImage.h>
    35 #include <geometry_msgs/PoseWithCovarianceStamped.h>
    36 #include <tf2_geometry_msgs/tf2_geometry_msgs.h>
    37                                                        

void ZEDWrapperNodelet::start_tracking() {
   754         NODELET_INFO_STREAM("*** Starting Positional Tracking ***");
   755         mNhNs.getParam("odometry_DB", mOdometryDb);
   756         mNhNs.getParam("pose_smoothing", mPoseSmoothing);
   757         mNhNs.getParam("spatial_memory", mSpatialMemory);
   758         mNhNs.getParam("floor_alignment", mFloorAlignment);
   759         mNhNs.getParam("init_odom_with_first_valid_pose", mInitOdomWithPose);
   760         NODELET_INFO_STREAM("Init Odometry with first valid pose data : " << (mInitOdomWithPose ? "ENABLED" : "DISABLED"));
   761 
   762         if (mInitialTrackPose.size() != 6) {
   763             NODELET_WARN_STREAM("Invalid Initial Pose size (" << mInitialTrackPose.size()
   764                                 << "). Using Identity");
   765             mInitialPoseSl.setIdentity();
   766 
   767             initTransforms();
   768         } else {
   769             set_pose(mInitialTrackPose[0], mInitialTrackPose[1], mInitialTrackPose[2],
   770                      mInitialTrackPose[3], mInitialTrackPose[4], mInitialTrackPose[5]);
   771         }
   772 
   773         if (mOdometryDb != "" && !sl_tools::file_exist(mOdometryDb)) {
   774             mOdometryDb = "";
   775             NODELET_WARN("odometry_DB path doesn't exist or is unreachable.");
   776         }
   777 
   778         // Tracking parameters
   779         sl::TrackingParameters trackParams;
   780         trackParams.area_file_path = mOdometryDb.c_str();
   781         trackParams.enable_pose_smoothing = mPoseSmoothing;
   782         NODELET_INFO_STREAM("Pose Smoothing : " << (trackParams.enable_pose_smoothing ? "ENABLED" : "DISABLED"));
   783         trackParams.enable_spatial_memory = mSpatialMemory;
   784         NODELET_INFO_STREAM("Spatial Memory : " << (trackParams.enable_spatial_memory ? "ENABLED" : "DISABLED"));
   785         trackParams.initial_world_transform = mInitialPoseSl;
   786 
   787         if (mFloorAlignment &&
   788             ((ZED_SDK_MAJOR_VERSION < 2) || (ZED_SDK_MAJOR_VERSION == 2 && ZED_SDK_MINOR_VERSION < 6))) {

and so on ;
Anyway I have change that code

Myzhar, 2018-09-28:
Have you updated to SDK v2.6?

guan4286, 2018-09-28:
my SDK's version is 2.4
Must it be updated to v2.6?

Myzhar, 2018-09-28:
If you  an update to 2.6 I suggest you to do it. The new SDK has a lot of improvements.

However the wrapper should compile also for SDK prior to v2.6, maybe a version check is missing somewhere.

guan4286, 2018-09-28:
Do you offer one-click upgrade

guan4286, 2018-09-28:
Isn't the ZED SDK downward compatible?

Myzhar, 2018-09-28:
The SDK v2.6 has new features that has been introduced in the ZED wrapper. The wrapper checks for the SDK version and enables the new features if available.
In particular SDK v2.6 introduces Floor Alignment, and a check is missing when enabling/disabling it

Myzhar, 2018-09-28:
To update the SDK download the correct version for you operating system here:
https://www.stereolabs.com/developers/release/2.6/

guan4286, 2018-09-28:
cuda need to update

Why not go down compatible

https://www.stereolabs.com/developers/release/2.6/

guan4286, 2018-09-28:
how to mute the thread

guan4286, 2018-09-29:
I already uninstall cuda9.1 and then install cuda9.2,why I still have this error report

https://www.stereolabs.com/developers/release/2.6/

Myzhar, 2018-09-29:
Make a `git pull` to get the latest updates published yesterday.
Then delete `build` and `devel` folders in your catlin workspace to clean the CMake cache.
Finally make a clean build with `catkin_make'

closed 2018-10-19 by Myzhar