stereolabs/zed-ros-wrapper #305: Incorrect covariances on the IMU messages

pvishal, 2018-10-04:
I use the ZED-M with the version `e736203` of this wrapper. The IMU messages I seem to be getting incorrect covariance values with on the first row populated. Here is what I see:
```
---
header:
  seq: 25409
  stamp:
    secs: 1538668854
    nsecs: 789258507
  frame_id: "base_link"
orientation:
  x: -0.0142320580781
  y: -0.00791106000543
  z: -0.0552176274359
  w: 0.998341560364
orientation_covariance: [4.39178991317749, 1.352308988571167, 0.30929282307624817, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: -0.00378964270524
  y: -0.0175797910747
  z: 0.00190697722691
angular_velocity_covariance: [0.019375164061784744, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
  x: 0.117462426424
  y: -0.0575326159596
  z: 10.2192306519
linear_acceleration_covariance: [0.0024830009788274765, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

Are those incorrect limits in the `for` loop on lines [1378](https://github.com/stereolabs/zed-ros-wrapper/blob/e736203c943ab0187cbb09ce371802dd0559b9b1/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1378) and [1416](https://github.com/stereolabs/zed-ros-wrapper/blob/e736203c943ab0187cbb09ce371802dd0559b9b1/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1416)?

--
Vishal

Myzhar, 2018-10-05:
Thank you for your report and for the fix. 
It's going to be merged.

pvishal, 2018-10-05:
Thanks. I have another question on the covariances (not sure if you prefer a new issue for this). The covariance matrix now look fine as below:

```
---
orientation:
  x: -0.0159692969173
  y: 0.0202658772469
  z: -0.0156502518803
  w: 0.999544560909
orientation_covariance: [4.263352394104004, 0.8899138569831848, 0.2573316693305969, 0.8899139761924744, 6.2962493896484375, 0.7558391094207764, 0.2573316693305969, 0.7558389902114868, 4.255410194396973]
---
```
The covariance values 4.26, 6.29 and 4.25 radians look quite high. Is that expected? Am I right in assuming they are radians?

Thanks!

Myzhar, 2018-10-05:
I must verify the units provided by the SDK. Maybe that this is related to the issue #300

PR #308 (2018-10-10): Fixed units for IMU covariance matrices for angular measures
Fixes #305
closed 2018-10-10 by adujardin