stereolabs/zed-ros-wrapper #307: Does ZED VO perform loop closure?

rhklite, 2018-10-07:
I noticed some drift when I was displaying the /path topic in rviz. I'm wondering if the Zed SDK's visual odometer performs loop closure?

Myzhar, 2018-10-07:
Yes, you must enable the `spatial_memory` option:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed_camera.launch#L105

to read more about it:
https://www.stereolabs.com/developers/documentation/API/v2.6.0/structsl_1_1TrackingParameters.html#abdcd6687ce52b670adf76eec8bee5fb4

rhklite, 2018-10-07:
Thank you for the reply.

Does the published /path & /odom estimation involves imu information aswell or if its purely visual odometry?

Myzhar, 2018-10-07:
The /path is generated by the complete Stereolabs SLAM algorithm.
The /odom is generated by the visual odometry algorithm and the IMU is used as prior to improve the orientation

rhklite, 2018-10-07:
Where might i find more papers/documentations on the Stereolabs SLAM algorithm?

Myzhar, 2018-10-07:
I'm sorry, but it is not open source.
All the available information are on the official documentation on www.stereolabs.com

rhklite, 2018-10-09:
Thank you for your help. I think I will close this thread now that my questions are answered.

closed 2018-10-09 by rhklite

rhklite, 2018-10-10:
Sorry, i realised i had more questions.

Do you know if the ZED Mini has a more accurate positional tracking over the ZED?

Myzhar, 2018-10-10:
The positional tracking of the ZED mini is improved by the inertial data that are fused to the visual tracking in the SLAM algortithm.
So, the reply is yes.
Furthermore, the ZED mini uses the "gravity" information to get the real orientation of the camera and is less afflicted by drift.

rhklite, 2018-10-10:
was there any benchmark done for Stereolab SLAM with ZED-M/ZED, such as KITTI/EUROC? Or any comparison against other popular SLAM algorithms such as ORBSLAM2/RTABMAP etc?