stereolabs/zed-ros-wrapper #311: Does zed-ros-wrapper support VIO topic ?

EvanYehh, 2018-10-18:
Hi
In the http://wiki.ros.org/zed-ros-wrapper
2.3 Published topics
Tracking
/zed/odom : Absolute 3D position and orientation relative to the odometry frame. (pure visual odometry)
/zed/map : Absolute 3D position and orientation relative to the map frame. (Sensor Fusion algorithm)
Inertial Data
/zed/imu/data : Accelerometer, Gyroscope and Orientation data in Earth frame.
/zed/imu/data_raw : Accelerometer and Gyroscope data in Earth frame.

It provides the VO topic/data through /zed/odom.
We wonder if zed can provide visual inertial odometry?
Or we need to fuse odom & imu with other package or method to get VIO topic/data

Thanks
Evan

WASCHMASCHINE, 2018-10-18:
Depends on your camera:
ZED = VO
ZED mini = VIO

Myzhar, 2018-10-18:
The Pose data of the ZED mini are always fused with the IMU data.
The IMU information is used as prior for each Visual Odometry step.

If you are using a ZED the IMU is not available and the Visual Odometry cannot be a VIO

EvanYehh, 2018-10-19:
Thanks for quick response.

By the way, In ZED mini which alg/method is applied for VIO?
Is filter-based or optimization-based, loosely-coupled or tightly-coupled ?

closed 2019-01-17 by Myzhar