stereolabs/zed-ros-wrapper #313: depth in decimeter with data type 'uint8' and 'mono8' encoding!!!!

JeyP4, 2018-10-23:
@Myzhar 

Hi
I am trying to get depth in decimeters(1m = 10dm).

Setting     `param name="max_depth"  value="25.5"`  ,  uint8(range 0-255) and 'mono8' encoding could be used.
I did respective changes in 'zed_wrapper_nodelet.cpp'.

But still depth map is coming in 'mono16' encoding after setting
`param name="openni_depth_mode"    value="true"`

Thank you fro consideration

[image]

Myzhar, 2018-10-23:
Hi @JeyP4,
your code seems correct. The fact that the depth is still `mono16` is weird.
Have correctly recompiled the code after the modifications?

I'm going to test it to see if there is something not visible at first sight. 

I'll let you know

JeyP4, 2018-10-23:
Wow, It worked.
Now I can save good amount of data to match allotted bandwidth.
Many Thanks
-Jai

Myzhar, 2018-10-23:
You are welcome

closed 2018-10-23 by Myzhar