stereolabs/zed-ros-wrapper #316: ROS Zed Camera Drift

Rafcin, 2018-10-25:
Hi, so I've been working on the zed camera for a while and I've been wondering how I can reduce the amount of drift I get from the camera.

So as of now, I've only calibrated the camera with my monitor in a dark room, but every time I train it, it still has issues with drift, and the depth map seems all crazy and stretchy on the sides, possibly i'm doing it wrong but I don't know.

I've calibrated it on a ubuntu desktop with a Titan XP and a Jetson Xavier and the results seem the same most of the time. I'm also wondering, I do have a Sparkfun IMU that I bought a while ago, am I able to merge that data with it while using ROS?

Myzhar, 2018-10-30:
A tutorial for the calibration process is coming. Actually you can make different tests until you reach a good configuration.

To merge the IMU data with the ZED data you can use an `UKF` or an `EKF`.
I suggest you to look at the `robot_localization` stack and to configure it to use the ZED odometry as visual odometry:
http://docs.ros.org/melodic/api/robot_localization/html/index.html#

Rafcin, 2018-10-31:
Can I use Matlab or the ROS calibration package? The issue th Zed camera goes insane when it look at the point cloud, nothing on it seems right, standard calibration through the app is ok, but the results never really change.

closed 2019-01-17 by Myzhar