stereolabs/zed-ros-wrapper #318: Error message about TF and image display issue

Zippen-Huang, 2018-10-27:
when I type command line<roslaunch zed_wrapper zed.launch>, and then <roslaunch ar_track_alvar ar_tag_track.launch>, here coming out a series of warning about TF, such as picture “error-TF”, but type <roslaunch zed_display_rviz display_zed.launch>, there is no warning message, I do not know why.
And sometimes, I subcribe topic “zed/left/image_raw” in RVIZ, no image come out, but when I change topic “zed/left/image_rect”, then I can view the video. I also do not know why.
[image]
[image]
[image]

Myzhar, 2018-10-29:
This is really weird. I will test it on TX2 to try to replicate the problem

Myzhar, 2018-10-29:
@Zippen-Huang I need to see your launch file to understand how you configured the system

Zippen-Huang, 2018-11-07:
if you do not come across the problem, can you send your code to me ?
I just follow the https://github.com/atomoclast/ar_tag_toolbox

Myzhar, 2018-11-12:
Please update to latest master version and look at the new example we added:
https://github.com/stereolabs/zed-ros-wrapper/tree/master/examples/zed_ar_track_alvar_example

Myzhar, 2018-11-19:
@Zippen-Huang any news about your problem? Can this issue be closed?