stereolabs/zed-ros-wrapper #321: reconfigure zed mini through ros wrapper

raymond30031, 2018-10-31:
Hello,

Is it possible to reconfigure the behavior of the zed mini through ros wrapper?
I would like to use the zed mini as a synchronized image and imu camera for reducing load on my computer.

Things I would like to reconfigure:
1. change the image raw output to bgr8 instead of bgra8. I just started working on computer vision, so could you tell me why you guys keep the alpha channel (for transparency I think) as part of the raw image?
2. Disabling tracking as i want to use it just like a camera, but still output ros messages for images and imu data

Thanks,
Juichung Kuo

raymond30031, 2018-11-01:
Also,
Is it possible to configure the imu hearder timestamp to  timestamp when it is recorded on the sensor and not to the image frame?

Myzhar, 2018-11-05:
Hi @raymond30031 

the "4 channel image" is a choice made at SDK level. I will check if it is possible to make something about it.

About the "tracking": it is started only if an external node subscribes to one of this topics:
- depth/depth_registered
- pose
- pose_with_covariance
- odom
- point_cloud/cloud_registered
- path_odom
- path_map
furthermore the parameter `depth_stabilization` must be set to false

About the IMU timestamp I'm going to add a parameter to set this feature.

Myzhar, 2018-11-06:
@raymond30031 Yes, if you use `sl::TIME_REFERENCE_CURRENT` you get the timestamp associated to the last IMU data received:
https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a453890423c830b627fc35c63edde7615

You can replace this line:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/src/nodelet/src/zed_wrapper_nodelet.cpp#L1383
with 
`imu_msg.header.stamp = sl_tools::slTime2Ros(imu_data.timestamp);`
to get the "real" timestamp associated with the IMU data.

About using the same timestamp for different IMU messages is due to a problem with TFs in ROS.

adujardin, 2018-11-06:
Hi @raymond30031
The GPU is used to convert the raw SbS YUV image to the rectified left + right RGB images.
Currently, the depth is also computed by default but you can disable it. Set the [`quality`](https://github.com/stereolabs/zed-ros-wrapper/blob/99cd29eb4a3489ec380a1485fda3fe1f0bf6f884/zed_wrapper/launch/zed_camera.launch#L91) parameter to `0` which is [`DEPTH_MODE_NONE`](https://www.stereolabs.com/docs/api/group__Depth__group.html#ga8d542017c9b012a19a15d46be9b7fa43).
The SDK should then have a minor impact on the GPU.

raymond30031, 2018-11-06:
@adujardin 
Thanks, setting quality to 0 cuts the memory usage in half.

@Myzhar 
I changed the frequency back to 500Hz for IMU becuz the documentation said you need 500 Hz to get all the data.

After changing the code as you suggested i get timestamps like this:
IMU:
seq: 2448
stamp: 
  secs: 0
  nsecs: 107068923
frame_id: "imu_link"

seq: 2449
stamp: 
  secs: 0
  nsecs: 107070912
frame_id: "imu_link"

seq: 2450
stamp: 
  secs: 0
  nsecs: 107072901
frame_id: "imu_link"

Left/Image:
seq: 81
stamp: 
  secs: 1541509515
  nsecs: 561439205
frame_id: "zed_left_camera_optical_frame"

seq: 82
stamp: 
  secs: 1541509515
  nsecs: 596072361
frame_id: "zed_left_camera_optical_frame"

seq: 83
stamp: 
  secs: 1541509515
  nsecs: 629726895
frame_id: "zed_left_camera_optical_frame"

It is apparent that the messages don't have the same time reference.
From the documentations:
[image timestamp](https://www.stereolabs.com/docs/api/2.7.0/classsl_1_1Camera.html#aeed4f708d46d12dd6b3c175ad076ba93)
[imu timestamp](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a453890423c830b627fc35c63edde7615)

For confirmation, the current implementation timestamps the images by the unix time that the image is loaded in memory. 
For the modified imu timestamp, it gives me the timestamp when the imu is received.
What exactly is referred when you say received? is it the unix time when the message exit the USB buffer?

Ideally, I would like to timestamp the messages (imu and images) with the clock onboard so they have the same reference. Is this possible?

Thanks,
JC

closed 2019-01-17 by Myzhar