stereolabs/zed-ros-wrapper #322: zed_rtabmap_example not working properly

griz1112, 2018-11-01:
I get no 3D Map. I suspect its something with tf. Run roswtf with this output.
ERROR TF multiple authority contention:
 * node [/rtabmap/rgbd_odometry] publishing transform [zed_camera_center] with parent [odom] already published by node [/zed/zed_wrapper_node]
 * node [/rtabmap/rtabmap] publishing transform [odom] with parent [map] already published by node [/zed/zed_wrapper_node]
 * node [/zed/zed_wrapper_node] publishing transform [zed_camera_center] with parent [odom] already published by node [/rtabmap/rgbd_odometry]
 * node [/zed/zed_wrapper_node] publishing transform [odom] with parent [map] already published by node [/rtabmap/rtabmap]
Also when using rqt to look at odom you see the frame changing from odom to zed_camera_center periodically. I assume I need to turn off odom in either zed_wrapper or rtabmap but so far haven't been able to do that and get the 3D map to work.

Myzhar, 2018-11-01:
The example has been designed to use only one TF. Maybe that something has been changed in Rtabmap with latest updates. 
I check it and I will fix the issue as soon as possible.

closed 2018-11-05 by Myzhar

Myzhar, 2018-11-05:
@griz1112 Dan, you can take the latest `master` version with the fixed example

PR #324 (2018-11-05): Fixed TF problems with RTABmap example
Fixes #322