stereolabs/zed-ros-wrapper #323: Frame rate drop while subscribing to image topics on Jetson TX2

Muskman, 2018-11-04:
Running zed camera on **Jetson TX2 ros kinetic**, **VGA** resolution in performance mode at **100fps**, I get a steady **50-53fps output**. However whenever I subscribe to any of the image topics the frame rate **_mysteriously drops to ~30fps_**. A similar thing happens when I am running at 15fps the frame rate drops to ~10-11fps.
_The frame rate jumps back up when the subscriber is killed_.
Has anyone come across this issue?

Myzhar, 2018-11-04:
@Muskman how do you measure the frame rate?

Muskman, 2018-11-04:
Using the command:  rosbag hz /zed/right/image_raw_color or for any other image topic by zed. When the images are not subscribed the frame rate is higher and then when there is a subscriber or if the topics are recorded the frame rate drops. Can't understand the reason behind it !

Myzhar, 2018-11-04:
I'm going to verify it on the TX2.
meanwhile can you verify if the board is set to maximum performances?
Try to see if all the clocks have maximum values with
`$ sudo ~/jetson_clocks.sh --show`

Myzhar, 2018-11-05:
I tested also at **720p@60FPS**:
[image]

still getting maximum framerate.

I have a new suggestion for you: when you compile the ROS wrapper, force the `Release` mode:
`$ catkin_make -DCMAKE_BUILD_TYPE=Release`

Muskman, 2018-11-06:
Okay, I updated the ZED SDK from 2.6 to 2.7 and built the recent zed wrapper using the `Release` mode. Now I am able to get 100fps in VGA mode. FPS problem solved ! 

**VGA 100 FPS**
[image]
_not so fast.. I also tested the following_

The problem when I subscribe to multiple topics published by the zed wrapper, the frame rate seems to drop down. I allocated a buffer size of 4096 `-b 4096` to record a bag with multiple topics that are shown in the bottom right terminal in the image below. 

**VGA 100FPS drops to ~60FPS**
[image]

The frame rate declines and saturates to around 60fps. Is it due to the 4GB memory allocation for recording the bag, i have tried with allocating the default buffer of 512 which also leads to a similar frame rate drop.

used the following command to record the bag :
`$ rosbag record -b 4096 /zed/depth/camera_info /zed/depth/depth_registered /zed/disparity/disparity_image /zed/left/camera_info /zed/left/image_raw_color /zed/odom /zed/path_map /zed/path_odom /zed/pose /zed/rgb/camera_info /zed/rgb/image_raw_color /zed/right/camera_info /zed/right/image_raw_color`

Myzhar, 2018-11-06:
The problem of the rosbag recording is known and it is not related to the wrapper, but to the available computational power.
For this reason we suggest to record instead an SVO using the ZED EXPLORER tool and then launch the ZED node using the SVO as input.

Muskman, 2018-11-06:
Did not know that recording a bag took up so much computation! That seems to be the case, thanks for the the constructive suggestions.

closed 2018-11-06 by Muskman