stereolabs/zed-ros-wrapper #326: jetson TX2 framedrop when image topic subscribed by remote machine

rhklite, 2018-11-08:
I'm trying to run RTABMAP with a ZED Mini Camera, with zed running on a Jetson TX2, publishing image_topics to a remote machine. I get frame drops when I subscribe to the image_topics from the remote machine, but I don't know how to improve my frame rate.

**Setup**

- I have a remote PC and a Nvidia Jetson TX2 that's installed with ZED SDK 2.7. The two machines are connected via a 1gigabit D-Link unmanaged network switch. Jetson is running ZED, with the remote machine subscribing to a image and viewing. 
- remote message publishing is done via the ros [multimaster_fkie](http://wiki.ros.org/multimaster_fkie) package. 

**Trouble Shooting Process**
- On Jetson, fps is checked with `rostopic hz /zed/rgb/image_rect`. The values shown 720HD@60fps/30fps, VGA@60fps/30fps was good, no frame drops. 
- On the remote machine, fps is also check with `rostopic hz /zed/rgb/image_rect_color`. When testing with VGA@60fps/30fps, frame rate is stable. But drops to 15/10 hz for 720HD@60fps/30fps. 
- On the remote machine, when both `rostopic hz /zed/rgb/image_rect` and `rosrun image_view image_view image:=/zed/rgb/image_rect_color` is running, the frame rate drops to ~15/10hz for any resolution/frame rate.

**My question**
1. Is this because of jetson limitation or network switch limitations?
2. Is there anything I could do to improve this on the zed_wrapper side?

Myzhar, 2018-11-09:
How are connected the Jetson and the remote machine?

If you are using the WiFi can you please make the same tests using a cable connection?

I confirmi you that your clock configuration is ok

adujardin, 2018-11-09:
@rhklite CPU 1 and 2 are offline (https://devtalk.nvidia.com/default/topic/1000345/jetson-tx2/two-cores-disabled-/)
To enable them run :

```bash
sudo su
echo 1 > /sys/devices/system/cpu/cpu1/online
echo 1 > /sys/devices/system/cpu/cpu2/online
~/jetson_clocks.sh
```

When the maximum performance mode is enabled you should see this :

```
SOC family:tegra186  Machine:quill
Online CPUs: 0-5
CPU Cluster Switching: Disabled
cpu0: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu1: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu2: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu3: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu4: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu5: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
GPU MinFreq=1300500000 MaxFreq=1300500000 CurrentFreq=1300500000
EMC MinFreq=40800000 MaxFreq=1866000000 CurrentFreq=1866000000 FreqOverride=1
Fan: speed=255
```

rhklite, 2018-11-11:
Thank you for the replies

@Myzhar The two machines are connected via a 1gigabit D-Link unmanaged network switch.
@adujardin I enabled the cpus as you described, though there is no improvement. 

 now the output of `~/jetson_clocks.sh --show` is

```
SOC family:tegra186  Machine:quill
Online CPUs: 0,3-5
CPU Cluster Switching: Disabled
cpu0: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu1: Gonvernor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=2035200
cpu2: Gonvernor=schedutil MinFreq=345600 MaxFreq=2035200 CurrentFreq=2035200
cpu3: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu4: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu5: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
GPU MinFreq=1122000000 MaxFreq=1122000000 CurrentFreq=1122000000
EMC MinFreq=40800000 MaxFreq=1600000000 CurrentFreq=1600000000 FreqOverride=1
Fan: speed=255
```

Myzhar, 2018-11-12:
`Online CPUs: 0,3-5`
The cores 1 and 2 are still offline
You will notice it also with the command `$ ~/tegrastats`

rhklite, 2018-11-13:
Sorry, I copy and pasted the wrong stats. The retest was done with core 1 and 2 online.

```
SOC family:tegra186  Machine:quill
Online CPUs: 0,3-5
CPU Cluster Switching: Disabled
cpu0: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu1: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu2: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu3: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu4: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
cpu5: Gonvernor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200
GPU MinFreq=1122000000 MaxFreq=1122000000 CurrentFreq=1122000000
EMC MinFreq=40800000 MaxFreq=1600000000 CurrentFreq=1600000000 FreqOverride=1
Fan: speed=255
```

Myzhar, 2018-11-13:
Still offline: 
`Online CPUs: 0,3-5`
you must get
`Online CPUs: 0-5`

rhklite, 2018-11-13:
you are right!

Though I dont think it complete solved my problem, but it's better than before now. Thank you!

closed 2018-11-14 by rhklite