stereolabs/zed-ros-wrapper #327: ZED Led: Blue indicator light on a multi-camera setup

sneheshs, 2018-11-10:
I am using 3 ZED cameras via 3 USB ports on my workstation and it is working fine. However, one issue is that the blue indicator LED seems unpredictable. I have all three cameras left and right rgb images along with depth images displaying fine in RVIZ. However, at the moment, at least only one of the lights are on. This can be misleading during data collection to check if the cameras are working or not. Can you please advise if there is a way to make sure the LEDs are on when they are working? BTW, when only one camera is being used all three of the cameras blue LED do come on, so there must be something with the multi-camera setup?

Please advise.

Myzhar, 2018-11-10:
I will forward the issue to the hardware developers and I will be back as soon as I get a reply.

Myzhar, 2020-03-23:
Solved with latest SDKs

closed 2020-03-23 by Myzhar