stereolabs/zed-ros-wrapper #328: Possibility of generating depth from bagfiles

aalapshah12297, 2018-11-11:
Hello,
I am using a ZED camera on a UAV. Since the UAV does not have good processing hardware (no NVIDIA GPU, and therefore no CUDA), I wish to use ZED SDK in post-processing.
More specifically, I plan to record a bagfile containing the raw stereo image in real time (which does not require the ZED SDK), and then I wish to use that bagfile later, on a different computer for generating depth, disparity, etc.

I am aware of other alternatives for doing so, which do not involve the use of the ZED SDK (such as [stereo_image_proc](http://wiki.ros.org/stereo_image_proc) and [depth_image_proc](http://wiki.ros.org/depth_image_proc)) but **I wish to know if it is possible to perform this using the ZED SDK itself**. By looking at the source code for the wrapper, I realized that the only certain camera parameters are passed to an `sl::Camera` object, which is then used to retrieve all data, _including the left and right images_. Instead, if there was some functionality in the SDK that allowed left and right images _to be passed to it_, and the corresponding depth image to be retrieved, then I could modify the wrapper for using it with my bagfile.

So is this currently a feature? If not, can you suggest an alternate setup (maybe with ROS running on a network of 2 computers) such that I can acquire images on one computer and process them on the other?

Myzhar, 2018-11-11:
@aalapshah12297 the ZED SDK does not accept images as input, so it is not possible to implement what you are asking for.
Furthermore recording the image stereo couples using rosbag is not a secure process. Indeed it is not assured that all the images of the stereo couples are correctly saved.
What you can do is to save an SVO using the "ZED Explorer" tool and you can post process it with the ZED wrapper that can that the SVO as input.

aalapshah12297, 2018-11-13:
Thanks for your response!
The alternative that you suggested seems like a good option. However, as I mentioned, my UAV does not have an NVIDIA gpu onboard. Is it possible to install ZED Explorer without CUDA installed?

Myzhar, 2018-11-13:
ZED Explorer does not require CUDA. You can force the installation of the SDK and use ZED Explorer to record the SVOs.

aalapshah12297, 2018-11-14:
I'll try this soon. You can close the issue. Thank you for your help :)

closed 2018-11-14 by Myzhar