stereolabs/zed-ros-wrapper #333: the frequency of the topic when I use rosbag to record /zed/right/image_raw_color /zed/left/image_raw_color

chenwinki, 2018-11-21:
I use rosbag record the topics /zed/right/image_raw_color /zed/left/image_raw_color, after a during , I found the count of left image is almost two times of the count of right image , what is the problem ,could you give me a hint? Thank you!
the following is the info of .bag file 
018-11-15-12-25-21.bag 
path:        mutil_camera_2018-11-15-12-25-21.bag
version:     2.0
duration:    17.2s
start:       Nov 15 2018 12:25:21.60 (1542255921.60)
end:         Nov 15 2018 12:25:38.79 (1542255938.79)
size:        2.6 GB
messages:    6885
compression: none [338/338 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu   [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /zed/imu/data_raw            6548 msgs    : sensor_msgs/Imu  
             /zed/left/image_raw_color     210 msgs    : sensor_msgs/Image
             /zed/right/image_raw_color    127 msgs    : sensor_msgs/Image

chenwinki, 2018-11-22:
Thank you for you reply ! I use PC. after I use another PC ,I did not meet the problem above, I think usb on my PC is bad.

chenwinki, 2018-11-22:
I have another question, I want to use three sensors : lidar imu camera,  in order to make sure they are synchronized，and I want add lidar driver in zed-ros-wrapper. could you give me some advice ? some references?

Myzhar, 2018-11-22:
My only advice is to not add the lidar code to the ZED wrapper. ROS manages synchronization using the topics timestamps.
Write your own node for the lidar or use its own if exists.
Then you can use `message_filters` to synchronize the topics.

You will find a lot of help about synchronization searching on answers.ros.org

chenwinki, 2018-11-23:
@Myzhar Thank you :+1: 
I have another question about the transformation from imu to left camera , I got two calibration results the Euler Angle from imu to left cam is [168.2  -86.7 -95]  and [ 45 -60 28]  in roll, pitch and yaw,  I think the roll and pitch is quite large assuming the PCB for the zed camera is flat. Do you have a probably result from imu to left camera?

Myzhar, 2018-11-23:
The angles should be very little. There was a bug in a previous version. Are you using the latest code available on `master` branch?

chenwinki, 2018-11-24:
oh! I use the previous version! why the latest version improve the frquency of imu from 500hz to 800hz？how can I slow down it?

Myzhar, 2018-11-24:
Because the firmware was improved and now it can read the IMU at maximum speed.
You can change the IMU frequency modifying the parameter in the file `zed_camera.launch` as described in the documentation:
https://www.stereolabs.com/docs/ros/zed_node/