stereolabs/zed-ros-wrapper #334: measure distance in Python

Zhefei777, 2018-11-23:
Hello, I want to measure the distance, but the Tutorial " zed_depth_sub_tutorial“ is in C++. Could anyone help me how to get the distance in Python？ 
Thank you

Myzhar, 2018-11-23:
Python examples are not currently planned, but you can follow the C++ tutorial replacing the Python code to subscribe to a topic as described in the ROS wiki:
http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29#rospy_tutorials.2BAC8-Tutorials.2BAC8-WritingPublisherSubscriber.Writing_the_Subscriber_Node

Zhefei777, 2018-11-26:
Thanks for your reply. I tried to write the python code, but I got all like this. 
�
�
�

�
Could you tell me what I did wrong. Here is my code:
`#! /usr/bin/env python2
# -*- coding: utf-8 -*-
import roslib
import sys
import rospy
import cv2
import numpy as np
import math
from cv_bridge import CvBridge, CvBridgeError
from geometry_msgs.msg import Twist
from sensor_msgs.msg import Image
from std_msgs.msg import Float32

class depth_processing():

    def __init__(self):

        rospy.init_node('zed_depth', anonymous=True)
        
        rospy.Subscriber("/zed/depth/depth_registered", Image, self.callback)
     
    
    def callback(self, depth):
        depths = depth.data
        # print(depth.data)
        #height, width = depths[0:2]
        
        #v = height/2
        #u = width/2
        center = 126524
        print(depths[center])

if __name__ == '__main__': 
    try:
        detector = depth_processing()
        rospy.spin()

    except rospy.ROSInterruptException:
        rospy.loginfo("Detector node terminated.")`

Zhefei777, 2018-11-29:
Thank you for your help！  It works！

closed 2018-12-07 by Myzhar

dipampatel18, 2020-07-15:
Thank you @Myzhar for the help! I observed that the depth obtained is quite inaccurate and keeps on fluctuating a lot. But, when I switched the 'u' and 'v' in depth_array[v, u], the depth obtained was pretty accurate. Can you think of any reason since the row and column are switched?