stereolabs/zed-ros-wrapper #335: the calibration between the imu and cameras using kalibr

chenwinki, 2018-11-25:
Dear author: :+1:
I using kalibr calibrate zed mini, I tried 20 times, But the timeoffset between imu and cameras is about 0.2 second, it is too large with imu: 800hz and camera:15hz, could you give me a hint? or the code of driver has a bug?

chenwinki, 2018-11-26:
This is the calibration procedure kalibr [https://github.com/ethz-asl/kalibr](url) ,I use this to calibrate stereo cameras, and imu between stereo cameras.The target used for kalibr 
[target.txt](https://github.com/stereolabs/zed-ros-wrapper/files/2614087/target.txt)

And I use imu_utils to calibrate imu to get the noise and bias , I got the result 
[imu_param.txt](https://github.com/stereolabs/zed-ros-wrapper/files/2614085/imu_param.txt)

and the calibration result 
[results-imucam-camimu800_2018-11-25-14-45-53.txt](https://github.com/stereolabs/zed-ros-wrapper/files/2614093/results-imucam-camimu800_2018-11-25-14-45-53.txt)

chenwinki, 2018-11-27:
@Myzhar  hello, could you give me some hint about how do you calibrate camera and imu, and the result.
You should have got accurate result, because people use zed sdk without providing the parameters.

Myzhar, 2018-11-27:
We are working on a tutorial on using Kalibr. Stay tuned ;)

chenwinki, 2018-11-27:
Waiting for your message！

Myzhar, 2018-11-27:
Here you are:
https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-

closed 2018-11-27 by Myzhar

chenwinki, 2018-11-28:
@Myzhar Thank you for your work!
1、but the timeoffset is still around 0.1s, this means the IMU and cameras is not  synchronized in the zed-ros-wrapper? when I use the imu and camera ,I should consider the timeoffset.
2、when I set the parameter imu_timestamp_sync in the file zed_camera.launch to true, this means the timestamp of some imu msgs will be same? 
hope for your reply!

Myzhar, 2018-11-28:
@chenwinki 
1. The IMU is faster than the camera, so a timeoffset is expected. The less the camera framerate, the more is the timeoffset. Furthermore you must think about USB buffering and other OS factors.
2. Exactly. An example: IMU 800Hz, camera 30 FPS. If you enable sync you will get at least 26 IMU messages with the same timestamp that is the timestamp of the latest grabbed image.

ZYJxxx, 2022-07-21:
@chenwinki hi, i saw your calibration result, translation between imu0 and cam0 is so small, it's so wired.

yhabib29, 2022-08-15:
@Myzhar Thank you for the tutorial !
Is there a reason why you use rectified images for calibration ?
Shouldn't we use raw images instead to fit a pinhole model ?

yhabib29, 2022-08-18:
Thanks @Myzhar, I managed to calibrate the stereo camera using Kalibr choosing pinhole-radtan model.
It does not provide repeatable results but the reprojection error is correct:
```
Calibration results 
====================
Camera-system parameters:
	cam0 (/cam0/image_raw):
	 type: <class 'aslam_cv.libaslam_cv_python.DistortedPinholeCameraGeometry'>
	 distortion: [-0.02699952 -0.00552966  0.00034329 -0.00139479] +- [ 0.00033004  0.00018569  0.00008031  0.00011824]
	 projection: [ 516.22781515  517.39320789  654.38520876  366.11877504] +- [ 0.29035704  0.28257418  0.33542257  0.23263578]
	 reprojection error: [0.000025, 0.000003] +- [0.758761, 0.682905]

	cam1 (/cam1/image_raw):
	 type: <class 'aslam_cv.libaslam_cv_python.DistortedPinholeCameraGeometry'>
	 distortion: [-0.02970289 -0.00364582 -0.00022852 -0.00121343] +- [ 0.00033202  0.00018969  0.00007939  0.00010378]
	 projection: [ 516.18737859  517.17068448  666.0365532   358.19751079] +- [ 0.28405359  0.27794324  0.32412273  0.22804654]
	 reprojection error: [-0.000023, -0.000001] +- [0.769643, 0.682531]

	baseline T_1_0:
	 q: [ 0.00102717  0.00054299  0.00085703  0.99999896] +- [ 0.0002903   0.00046198  0.00006375]
	 t: [-0.1198369  -0.00012308  0.00085938] +- [ 0.00004663  0.00004524  0.00009093]

Target configuration
====================

  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.088 [m]
    Spacing 0.0264 [m]
```

About the IMU parameters, can you confirm I can use noise density and random walk parameters returned by ZED SDK:
- Gyroscope noise density:          0.007 deg/s/√Hz (TO BE CONVERTED 
- Gyroscope random walk:          0.0019474 deg/s²/√Hz
- Accelerometer noise density:    0.0016 m/s²/√Hz
- Accelerometer random walk:     0.0002509 m/s³/√Hz

Which leads to the following Kalibr configuration file:
```
#Accelerometers
accelerometer_noise_density: 1.6e-03   #Noise density (continuous-time)
accelerometer_random_walk:   2.509e-4  #Bias random walk

#Gyroscopes
gyroscope_noise_density:     1.2217304763960306e-4  #Noise density (continuous-time)
gyroscope_random_walk:       3.398854185333757e-05  #Bias random walk

rostopic:                    /imu/data_raw     #IMU ROS topic
update_rate:                 400.0             #Hz (for discretization of the values above)
```

mandelyoni, 2023-04-19:
Dear Sir,

I'm using Zed2i's sensors (left camera and IMU) with an external monocular VIO pipeline. The extrinsics I receive from the ROS topic /left_cam_imu_transformation and from Kalibr are very different.

The extrinsics from /left_cam_imu_transformation is:
rostopic echo /zed2i/zed_node/left_cam_imu_transform
translation:
x: -0.0020000000949949026
y: -0.023000003769993782
z: 0.0002200000308221206
rotation:
x: -0.0010500885546207428
y: -0.0
z: -0.00393286207690835
w: 0.9999917149543762
which is equivalent to -
[0.999969, 0.007864, -0.000099, -0.002000,
-0.007864, 0.999967, 0.002121, -0.023000,
0.000116, -0.002120, 0.999998, 0.000220,
0, 0, 0, 1]

While the extrinsics from Kalibr (https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-) are :
cam0:
T_cam_imu:

[-0.004248117180815025, -0.9998625130142822, -0.016028367639358054, 8.41951564249623e-05]
[0.001063559295260974, 0.0160239856188632, -0.9998710420481793, -1.708597954171419e-06]
[0.9999904111250486, -0.004264616471714999, 0.0009953413010867496, 3.811715513237331e-06]
[0.0, 0.0, 0.0, 1.0]
cam_overlaps: [1]
camera_model: pinhole
distortion_coeffs: [0.0006582543326556634, -0.001338594740114625, 0.0003683086103016461,
0.0016571045464767074]
distortion_model: radtan
intrinsics: [662.3788576876452, 662.6400636208534, 654.9267980662693, 366.0819275708851]
resolution: [1280, 720]
rostopic: /zed/zed_node/left/image_rect_color
timeshift_cam_imu: 0.10015660064007245 (also strange)
In addition, while the camera is horizontal and facing forward, the output of /zed2i/zed_node/imu/data_raw is:
header:
seq: 17969
stamp:
secs: 1681195935
nsecs: 453395218
frame_id: "zed2i_imu_link"
orientation:
x: 0.0
y: 0.0
z: 0.0
w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
x: -0.058271501732520346
y: -0.1356413144708724
z: 0.06934011135360624
angular_velocity_covariance: [6.595000957109997e-10, 0.0, 0.0, 0.0, 1.9613346042712762e-09, 0.0, 0.0, -0.0, 7.858832729846918e-10]
linear_acceleration:
x: 1.5133049488067627
y: 0.3734425902366638
z: 9.75547981262207
linear_acceleration_covariance: [0.007840245962142944, 0.0, 0.0, 0.0, 0.008192301727831364, 0.0, 0.0, -0.0, 0.007778662722557783]

You can see the z axis of linear_acceleration is pointing up. According to /left_cam_imu_transform it should be pointing forward (like the camera).

Please advise.

Thank you, Yoni

Steps to Reproduce
rostopic echo /zed2i/zed_node/left_cam_imu_transform
Follow https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-
rostopic echo /zed2i/zed_node/imu/data_raw
Expected Result
Transformation should look similar

Actual Result
Transformation not similar

ZED Camera model
ZED2i

Environment
ROS Noetic on Ubuntu 20.0 on an intel chip (Lenovo Thinkpad)

Myzhar, 2023-04-19:
@mandelyoni please do not cross-post (https://github.com/stereolabs/zed-ros-wrapper/issues/335#issuecomment-1514162061) and do not reply to closed issues, it's preferred to open a new issue and cite the closed one if you need it.