stereolabs/zed-ros-wrapper #339: save depth image camera zed

abdo-robotic, 2018-12-05:
how i can save depth image from /zed/depth/depth_registered  topic

Myzhar, 2018-12-05:
Do you want to save every single frame in a PNG or JPG image or do you want to save a stream that ca be used by ROS?

abdo-robotic, 2018-12-05:
i want to take some screenshot  in a png or jpg

Myzhar, 2018-12-05:
If you need only some screenshoots you can follow this simple procedure:
1) Start the node: `$ roslaunch zed_wrapper zed.launch`
2) Start rqt_view: `$ rqt_image_view`
3) Select the correct topic in the top left list box
4) Click the top right button to save screenshoots

It `rqt_image_view` is not available you can install it using
`$ sudo apt install ros-${ROS_DISTRO}-rqt-image-view`

closed 2018-12-07 by Myzhar

zoombinis, 2020-12-10:
@Myzhar I followed these steps and they were very helpful, thank you. But how is the depth information still accessible from the saved .png screenshot image?

Are there any simple C++ examples for how to denormalize the [0 - 255] "depth normalization" without using the zed SDK?

Myzhar, 2020-12-10:
@zoombinis if you need a "de-normalized" depth simply save it without performing normalization.
To de-normalize the depth you need information (min depth, max depth) that are not available in a normalized PNG image

zoombinis, 2020-12-10:
@Myzhar Thanks, but if I understand correctly that means saving the raw array of floats to a text file since any depth image displayed by `rqt_image_view` has already been normalized, correct?

Myzhar, 2020-12-10:
I do not know how `rqt_image_view` internally works, but I suppose that it normalizes the image and it saves exactly what's rendered on screen.
To save the "raw" float depth I suggest you write a new node that subscribes to the depth topic and saves it.

PS please do not reply to a closed issue. It's normally better to open a new one and cite the closed if you need. In this way, it's easier to track new questions and eventual new bugs.