stereolabs/zed-ros-wrapper #340: reset zed camera odometry

abdo-robotic, 2018-12-07:
I'm  using zed camera to get my robot position ( http://wiki.ros.org/zed-ros-wrapper ) subscribing to /odom topic .  during the navigation of the robot and after it gets to one point  i want to reset  publishing values of the /odom to "0 0 0 0 0 0 " to continue the navigation and have a good precision of the next trajectory .

Myzhar, 2018-12-07:
Reading the documentation you can find that a ROS service exists that does exactly what you are asking for, `reset_odometry`:
https://www.stereolabs.com/docs/ros/zed_node/#services

You can call the service using the `rqt` GUI or by command line:
`$ rosservice call /zed/reset_odometry`

abdo-robotic, 2018-12-07:
okay thank you i will see how i can add it to my code cause the robot navigate autonomously

Myzhar, 2018-12-07:
You can call a service by code:
http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28c%2B%2B%29#roscpp_tutorials.2BAC8-Tutorials.2BAC8-WritingServiceClient.Writing_the_Client_Node

abdo-robotic, 2018-12-07:
yeah i know thank you sire

abdo-robotic, 2018-12-07:
i get this error 
ERROR: Service [/zed/reset_odometry] is not available.
i  called the service from terminal using this line : rosservice call /zed/reset_odometry

Myzhar, 2018-12-07:
You can see the list of available services using the command:
`$ rosservice list`

Are you using the latest version of the wrapper available in the master branch?

abdo-robotic, 2018-12-07:
how i can get the last version without removing cuda or  somethings else

Myzhar, 2018-12-07:
If you are using CUDA 9.0 you must not remove it to update to the latest version.
Take the latest SDK v2.7 for CUDA 9 (https://www.stereolabs.com/developers/release/) and follow the installation instruction here:
https://www.stereolabs.com/docs/ros/#installation

abdo-robotic, 2018-12-07:
yeah i have CUDA 9.0  
do i have to remove the old ros wrapper package !?.?

Myzhar, 2018-12-07:
If you installed it using GIT you only need to issue a `$ git pull` command in the root of the wrapper and then issue the `$ catkin_make` command in the root of your catkin workspace to compile it again

closed 2019-01-02 by Myzhar