stereolabs/zed-ros-wrapper #345: The rosbag record problem

sunmiaobo, 2018-12-12:
Hi author,
I used ros to record those three topic zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data.The terminal showed "buffsize exceeded".
Firstly, I try to use all the momory to record the three topic,but the 16GB memory can't support the 10 mins.
Secondly,I used the topic_tools throttle message to set the the right and left topics' publishing frequency .About 10 hz the ros can deal the message which buffer don't exceed.
I want to get the the right camera ,left camera and imus' synchronization time data .How can I get those.
[image]
[image]
[image]

Myzhar, 2018-12-12:
Hi,
please give us more information about your system (Nvidia Jetson, PC Desktop, Laptop, ecc).

Meanwhile you can read more on this old issue:
https://github.com/stereolabs/zed-ros-wrapper/issues/333

sunmiaobo, 2018-12-12:
I use Thinkpad laptop T480.
[image]
And PC Desktop 
[image]
The laptop is installed the cuda 10 and the Desktop is installed the cuda 9.Both of them installed the Ubuntu 16.04 and Ros kinetic.

AdamPengG, 2018-12-14:
Hi Myzhar,
Would I redefine and summarize my questions.
I used rosbag to record those three topic zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data.The terminal showed "buffsize exceeded".Pic 1 shows that.
Firstly, I try the ros commond "rosbag record -o zed zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data --buffsize=0 " use all the momory to record the three topic,but the 16GB memory can't support the 10 mins.When I record 3 topics the Desktop's memory reduce fast.
As the Pic 2 and Pic 3 shows.
Secondly,I used the topic_tools throttle message to set the the right camera raw pic and left camera raw pic topics' publishing frequency.I have tried the 25Hz,20Hz,15Hz and 10Hz.And about 10 Hz the ros can deal the message which buffer don't exceed.
I will show you my Laptop's and Desktop's Configuration.Pic 4 is Laptop Pic 5 is Desktop.
[image]
[image]
[image]
[image]
[image]

Myzhar, 2018-12-14:
Please stop opening new issues

closed 2018-12-14 by Myzhar

hanzheteng, 2019-12-27:
@sunmiaobo @AdamPengG From my guess, the issue is that your local disk cannot offer fast enough writing speed. (probably not SSD?) 

Given the current computer configuration, The walk around solutions could be 1) reduce the resolution of camera frame, 2) reduce the frame rate, or 3) reduce the number of topics to be recorded at the same time. The main objective here is to reduce the throughput to be written on your disk.

Another walkaround could be using `rosbag record -b 2048 <topic name>` to extend the buffer size. However, this can only make the warning happen later (probably double the amount of time) but cannot fix the problem. Also, you have to wait for a while after you terminate the rosbag command, and make sure the `xxx.bag.active` file has been turned into `xxx.bag` file.