stereolabs/zed-ros-wrapper #346: ROSBag record problem

AdamPengG, 2018-12-14:
Hi author,
I used rosbag to record those three topic zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data.The terminal showed "buffsize exceeded".Pic 1 shows that.
Firstly, I try the ros commond "rosbag record -o zed zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data --buffsize=0 " use all the momory to record the three topic,but the 16GB memory can't support the 10 mins.When I record 3 topics the Desktop's memory reduce fast.
As the Pic 2 and Pic 3 shows.
Secondly,I used the topic_tools throttle message to set the the right camera raw pic and left camera raw pic topics' publishing frequency.I have tried the 25Hz,20Hz,15Hz and 10Hz.And about 10 Hz the ros can deal the message which buffer don't exceed.
I will show you my Laptop's and Desktop's Configuration.Pic 4 is Laptop Pic 5 is Desktop.
[image]
[image]
[image]
[image]
[image]

I want to get the the 30Hz synchronous right camera raw pic,left camera raw pic  and 100Hz imu data .How can I get them.

Myzhar, 2018-12-14:
Please do not duplicate issues #345

closed 2018-12-14 by Myzhar

AdamPengG, 2018-12-14:
Sorry Myzhar. I have just signed in a new git.T  The issues is 345 is my partner's git.I want to make the problem clearer.So I edit a new issue.Hope your response.