stereolabs/zed-ros-wrapper #347: Rosbag key problem.

AdamPengG, 2018-12-14:
Hi Myzhar,
Would I redefine and summarize my questions.
I used rosbag to record those three topic zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data.The terminal showed "buffsize exceeded".Pic 1 shows that.
Firstly, I try the ros commond "rosbag record -o zed zed/right/image_raw_color，zed/left/image_raw_color and zed/imu/raw_data --buffsize=0 " use all the momory to record the three topic,but the 16GB memory can't support the 10 mins.When I record 3 topics the Desktop's memory reduce fast.
As the Pic 2 and Pic 3 shows.
Secondly,I used the topic_tools throttle message to set the the right camera raw pic and left camera raw pic topics' publishing frequency.I have tried the 25Hz,20Hz,15Hz and 10Hz.And about 10 Hz the ros can deal the message which buffer don't exceed.

I will show you my Laptop's and Desktop's Configuration.Pic 4 is Laptop Pic 5 is Desktop.
And the issue #333 isn't useful. .Hope your response.
I'm sorry to disturb you.

[image]
[image]
[image]
[image]
[image]

Myzhar, 2018-12-14:
We are going to check your problem, but please stop opening new issues about the same problem.

AdamPengG, 2018-12-15:
Thanks Myzhar

AdamPengG, 2019-01-07:
Is anybody here?

Myzhar, 2019-01-07:
We could not replicate your problem. Since it seems not to be related to the ZED wrapper, I suggest you to search for similar questions (or to post a new one) on the ROS Answers platform: https://answers.ros.org/questions/

A few suggestions: check for disk speed, buffer sizes and other possible operating system causes.

closed 2019-01-07 by Myzhar

hanzheteng, 2019-12-27:
This is not the problem of zed camera, but a general issue on memory access. (Though we had the similar issue today on zed camera.) The speed you are trying to write on the disk is faster than the maximum writing speed of the disk itself. Therefore, as time goes on, the data queued in the buffer becomes more and more and eventually overwhelms. See