stereolabs/zed-ros-wrapper #348: Use the ros wrapper to record .svo files?

axjensen, 2018-12-15:
Hi,
Is it possible to use the ROS wrapper to record a .svo file?
I was wondering, if I could use ROS + multiple computers with each one ZED camera attached, to record .svo files on each computer+zed instance.
I was thinking a single computer acted as master node and all other computer+ZED nodes where told to record by the master.
This way I could collect .svo files from many cameras.
I am aware of the multi-camera recording solution, but for my approach, I can't attach multiple cameras to a single computer.

Myzhar, 2018-12-15:
Hi, this feature is not currently available, but it could be possible to add a service to start the SVO recording and one to stop it.
We were just planning to add something similar and your issue could be the right occasion to do it.
Stay tuned 😉

axjensen, 2018-12-15:
Hi @Myzhar 
Thank you very much for your quick answer and thank you for considering features like this.
Just to give you some background information on why I am asking (I am not sure you can use that information):
I have multiple Jetson TX2's on a local network. Each JTX2 has a single ZED camera attached. On the network, there is a central computer which I refer to as the ROS master node. This master node should notify all the JTX2 nodes to record and eventually stop recording. When recording is done, the master node should be able to get information about the JTX2 nodes recorded .svo files. For instance if the master wants to copy the .svo file from the JTX2 nodes to it self.
I will wait patiently for any updates ;-)

mikedef, 2018-12-28:
Hi @Myzhar 
I am very much interested in the ability to record SVO files via the ROS wrapper. My application is with a single camera, but would like to be able to enable recording of the image streams through the command line or a launch file. 
Thank you,
-Mike

Myzhar, 2019-01-31:
Features added with https://github.com/stereolabs/zed-ros-wrapper/pull/367

closed 2019-01-31 by Myzhar