stereolabs/zed-ros-wrapper #349: update zed sdk from v2.3 to 2.7

abdo-robotic, 2018-12-17:
i want to update my zed-ros-wrapper package  to get the new one 
i have cuda 9.0 sdk v2.3 ubuntu 16.04
how i can update zed sdk to v2.7   to get the new zed-ros-wrapper package

Myzhar, 2018-12-17:
You can download the SDK v2.7 for Ubuntu 16 and CUDA 9 from this page: https://www.stereolabs.com/developers/release/
Scroll down and you will find it

To update the wrapper simply go to your catkin workspace then
`$ cd src/zed-ros-wrapper`
`$ git pull`
finally go back to your catkink workspace root and compile:
`$ catkin_make`

abdo-robotic, 2018-12-17:
after i run $git pull 
i got this error 
fatal: Not a git repository (or any of the parent directories): .git

Myzhar, 2018-12-17:
How did you install the ZED wrapper?
`git pull` works only if you have installed it from source making a `git clone` of the zed-ros-wrapper repository as explained here:
https://www.stereolabs.com/docs/ros/#installation

abdo-robotic, 2018-12-17:
i remove the old one and i followed this lines:
cd ~/catkin_ws/src
git clone https://github.com/stereolabs/zed-ros-wrapper.git
cd ../
catkin_make
and i got :+1: 
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find CUDA: Found unsuitable version "9.0", but required is exact
  version "10" (found /usr/local/cuda-9.0)
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:386 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindCUDA.cmake:949 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:51 (find_package)
  zed-ros-wrapper/zed_wrapper/CMakeLists.txt:22 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/abanay/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/abanay/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

abdo-robotic, 2018-12-18:
i solved the problems but i got this : 

roslaunch zed_wrapper zed.launch 
... logging to /home/abanay/.ros/log/45139b06-02aa-11e9-8543-548ca0efeab7/roslaunch-abanay-X556UV-7739.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://abanay-X556UV:43725/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_wrapper_node/auto_exposure: True
 * /zed/zed_wrapper_node/base_frame: zed_camera_center
 * /zed/zed_wrapper_node/camera_flip: False
 * /zed/zed_wrapper_node/camera_model: 1
 * /zed/zed_wrapper_node/confidence: 100
 * /zed/zed_wrapper_node/confidence_img_topic: confidence/confid...
 * /zed/zed_wrapper_node/confidence_map_topic: confidence/confid...
 * /zed/zed_wrapper_node/depth_cam_info_topic: depth/camera_info
 * /zed/zed_wrapper_node/depth_stabilization: 1
 * /zed/zed_wrapper_node/depth_topic: depth/depth_regis...
 * /zed/zed_wrapper_node/disparity_topic: disparity/dispari...
 * /zed/zed_wrapper_node/exposure: 100
 * /zed/zed_wrapper_node/floor_alignment: False
 * /zed/zed_wrapper_node/frame_rate: 30
 * /zed/zed_wrapper_node/gain: 100
 * /zed/zed_wrapper_node/gpu_id: -1
 * /zed/zed_wrapper_node/imu_frame: imu_link
 * /zed/zed_wrapper_node/imu_pub_rate: 800.0
 * /zed/zed_wrapper_node/imu_timestamp_sync: True
 * /zed/zed_wrapper_node/imu_topic: imu/data
 * /zed/zed_wrapper_node/imu_topic_raw: imu/data_raw
 * /zed/zed_wrapper_node/init_odom_with_first_valid_pose: True
 * /zed/zed_wrapper_node/initial_tracking_pose: [0.0, 0.0, 0.0, 0...
 * /zed/zed_wrapper_node/left_cam_info_raw_topic: left/camera_info_raw
 * /zed/zed_wrapper_node/left_cam_info_topic: left/camera_info
 * /zed/zed_wrapper_node/left_camera_frame: zed_left_camera_f...
 * /zed/zed_wrapper_node/left_camera_optical_frame: zed_left_camera_o...
 * /zed/zed_wrapper_node/left_raw_topic: left/image_raw_color
 * /zed/zed_wrapper_node/left_topic: left/image_rect_c...
 * /zed/zed_wrapper_node/mat_resize_factor: 1.0
 * /zed/zed_wrapper_node/max_depth: 10.0
 * /zed/zed_wrapper_node/odometry_db: 
 * /zed/zed_wrapper_node/odometry_frame: odom
 * /zed/zed_wrapper_node/odometry_topic: odom
 * /zed/zed_wrapper_node/openni_depth_mode: 0
 * /zed/zed_wrapper_node/path_max_count: -1
 * /zed/zed_wrapper_node/path_pub_rate: 2
 * /zed/zed_wrapper_node/point_cloud_topic: point_cloud/cloud...
 * /zed/zed_wrapper_node/pose_frame: map
 * /zed/zed_wrapper_node/pose_smoothing: False
 * /zed/zed_wrapper_node/pose_topic: pose
 * /zed/zed_wrapper_node/publish_map_tf: True
 * /zed/zed_wrapper_node/publish_pose_covariance: False
 * /zed/zed_wrapper_node/publish_tf: True
 * /zed/zed_wrapper_node/quality: 1
 * /zed/zed_wrapper_node/resolution: 2
 * /zed/zed_wrapper_node/rgb_cam_info_raw_topic: rgb/camera_info_raw
 * /zed/zed_wrapper_node/rgb_cam_info_topic: rgb/camera_info
 * /zed/zed_wrapper_node/rgb_raw_topic: rgb/image_raw_color
 * /zed/zed_wrapper_node/rgb_topic: rgb/image_rect_color
 * /zed/zed_wrapper_node/right_cam_info_raw_topic: right/camera_info...
 * /zed/zed_wrapper_node/right_cam_info_topic: right/camera_info
 * /zed/zed_wrapper_node/right_camera_frame: zed_right_camera_...
[... 48 lines cut ...]
[ WARN] [1545126597.543638098]: Camera model does not match user parameter. Please modify the value of the parameter 'camera_model' to 0
[ INFO] [1545126597.543887610]: CAMERA MODEL : ZED
[ INFO] [1545126597.715361411]: Advertised on topic rgb/image_rect_color
[ INFO] [1545126597.777315613]: Advertised on topic rgb/image_raw_color
[ INFO] [1545126597.841995652]: Advertised on topic left/image_rect_color
[ INFO] [1545126597.905079178]: Advertised on topic left/image_raw_color
[ INFO] [1545126597.939727554]: Advertised on topic right/image_rect_color
[ INFO] [1545126597.969006885]: Advertised on topic right/image_raw_color
[ INFO] [1545126598.000336171]: Advertised on topic depth/depth_registered
[ INFO] [1545126598.033157052]: Advertised on topic confidence/confidence_image
[ INFO] [1545126598.034032936]: Advertised on topic confidence/confidence_map
[ INFO] [1545126598.035185122]: Advertised on topic disparity/disparity_image
[ INFO] [1545126598.036424665]: Advertised on topic point_cloud/cloud_registered
[ INFO] [1545126598.037507570]: Advertised on topic rgb/camera_info
[ INFO] [1545126598.038348531]: Advertised on topic left/camera_info
[ INFO] [1545126598.039161985]: Advertised on topic right/camera_info
[ INFO] [1545126598.040123970]: Advertised on topic depth/camera_info
[ INFO] [1545126598.040978320]: Advertised on topic rgb/camera_info_raw
[ INFO] [1545126598.041809403]: Advertised on topic left/camera_info_raw
[ INFO] [1545126598.042638174]: Advertised on topic right/camera_info_raw
[ INFO] [1545126598.043395535]: Advertised on topic pose
[ INFO] [1545126598.044183508]: Advertised on topic odom
[ INFO] [1545126598.044984280]: Advertised on topic path_odom
[ INFO] [1545126598.046149317]: Advertised on topic path_map
[ WARN] [1545126598.046310832]: 'imu_pub_rate' set to 800 Hz but ZED camera model does not support IMU data publishing.

abdo-robotic, 2018-12-18:
[ INFO] [1545126930.297978494]: *** Starting Positional Tracking ***
[ INFO] [1545126930.300854563]: Init Odometry with first valid pose data : ENABLED
[ INFO] [1545126930.300934151]: Pose Smoothing : DISABLED
[ INFO] [1545126930.300976504]: Spatial Memory : DISABLED
[ INFO] [1545126930.301017548]: Floor Alignment : DISABLED
[ INFO] [1545126930.386052779]: Tracking ENABLED
[ INFO] [1545126930.499998090]: Odometry aligned to last tracking pose
[ WARN] [1545127081.241964853]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[ WARN] [1545127091.242165402]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[ WARN] [1545127101.242373837]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
[ WARN] [1545127111.536891518]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.
^C[zed/zed_state_publisher-3] killing on exit
[zed/zed_wrapper_node-2] killing on exit
^C^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Myzhar, 2018-12-18:
You are using a ZED, so the IMU data are not available (only "ZED mini" has an IMU inside), so that warning is correct. You can ignore it or you can modify `zed.launch`, setting `camera_model` to 0, as suggested a few line above.

The warning about FPS is normal when the node starts. If you continue to get it during the running it means that your GPU card is not powerful enough for the "resolution+FPS" settings that you are using.
You can try to reduce resolution and/or FPS to get better performances

abdo-robotic, 2018-12-18:
what about this warning :+1: [ WARN] [1545127847.427291770]: Elaboration takes longer than requested by the FPS rate. Please consider to lower the 'frame_rate' setting.

Myzhar, 2018-12-18:
As I said:

> The warning about FPS is normal when the node starts. If you continue to get it during the running it means that your GPU card is not powerful enough for the "resolution+FPS" settings that you are using.
You can try to reduce resolution and/or FPS to get better performances

abdo-robotic, 2018-12-18:
okay thank you so much last question please what this means :/
 rosservice info /zed/reset_odometry 
Node: /zed/zed_wrapper_node
URI: rosrpc://abanay-X556UV:51479
Type: zed_wrapper/reset_odometry
ERROR: Unable to load type [zed_wrapper/reset_odometry].
Have you typed 'make' in [zed_wrapper]?

closed 2018-12-18 by Myzhar