stereolabs/zed-ros-wrapper #350: reset_odometry call error

abdo-robotic, 2018-12-18:
when i call reset_odometry i get this feedback
rosservice call /zed/reset_odometry 
ERROR: Unable to load type [zed_wrapper/reset_odometry].
Have you typed 'make' in [zed_wrapper]?

Myzhar, 2018-12-18:
Have you recompiled the wrapper after the update?

abanay, 2018-12-18:
i got it thanks 
i can call the reset buy command but by code i still got problem
#! /usr/bin/env python

import rospy
from zed_wrapper.srv import reset_tracking
import sys

rospy.init_node('client')
reset_simulation = rospy.ServiceProxy('zed_wrapper/reset_tracking', reset_tracking)
print reset_simulation

abdo-robotic, 2018-12-18:
i got it thank you 
but i have problem with the code

#! /usr/bin/env python

import rospy
from zed_wrapper.srv import reset_tracking
import sys

rospy.init_node('client')
reset_simulation = rospy.ServiceProxy('zed_wrapper/reset_tracking', reset_tracking)
print reset_simulation

abdo-robotic, 2018-12-18:
what i miss in my code:

 #! /usr/bin/env python

import rospy
from zed_wrapper.srv import reset_tracking
import sys

rospy.init_node('client')
rospy.ServiceProxy('/zed/reset_tracking', reset_tracking)

abdo-robotic, 2018-12-18:
it's okay thank you

closed 2018-12-18 by Myzhar