stereolabs/zed-ros-wrapper #352: exit from while not rospy.is_shutdown()  loop without completely shutting down the node

abdo-robotic, 2018-12-20:
how i can exit from while not rospy.is_shutdown()  loop without completely shutting down the node

this is my code ( exit from the loop and function() )

def function():
   ..
   while not rospy.is_shutdown():
          ..  
           ..

def clbk_laser(msg):
    ..
    .. 
    function()
    ..
    ..

def main():
    rate = rospy.Rate(20)
    while not rospy.is_shutdown():
              ..  
              ..
...
...
if __name__ == '__main__':
    main()

Myzhar, 2018-12-20:
Sorry, but I do not really understand what you are trying to do.

abdo-robotic, 2018-12-20:
inside a function i am using while not rospy.is_shutdown() loop inside a function  
what i need is a way to exit from the loop and the function without  shutting down the node

abdo-robotic, 2018-12-20:
i tested  many ways like  exit  kill the node return and break do nothing

Myzhar, 2018-12-20:
It's a ROS/Python question, you should check on ROS answer website or post the question there.

This is not the right place

closed 2018-12-20 by Myzhar

abdo-robotic, 2018-12-20:
ok thank you